jareware/chilipie-kiosk #5: Controlled reboot

jareware, 2016-04-20:
Currently, the recommended way to shut down the Pi is actually to just unplug it.

For some reason, when you go `$ sudo reboot` via SSH/terminal access, it'll briefly bring up Chromium again after the shutdown's started, and it will subsequently lose its open tabs setup.

This shouldn't happen.

jareware, 2016-04-20:
Actually, it's probably this: https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1322895.html

jareware, 2018-08-17:
Not an issue anymore as of the [2.0.0 release](https://github.com/futurice/chilipie-kiosk/releases/tag/v2.0.0).

closed 2018-08-17 by jareware

Elvincth, 2018-11-30:
Sorry , how to disable auto reboot

jareware, 2018-12-03:
Hey @Elvincth, please check out the instructions in the [first boot doc](https://github.com/futurice/chilipie-kiosk/blob/master/docs/first-boot.md#automating-things)!