jareware/chilipie-kiosk #9: chromium-browser does not work on Google sites

kaplas, 2016-06-02:
Google sites, such as Google Analytics dashboards, do not work with the version of `chromium-browser` currently installed. Attempts to access those pages give the famous `_ERR_SSL_PROTOCOL_ERROR_` and the `--ignore-certificate-errors` does not affect the situation.

According to [an answer at askubuntu.com](http://askubuntu.com/a/737053) this issue manifested in some particular versions of `chromium-browser` package, and have since been fixed. So an update to the package would be needed here. However, I heard from @jareware that more current version of Chromium segfaults on the image at the moment, so there might be even more work required to fix this issue than what it looks like.

kimmobrunfeldt, 2016-08-31:
I think we are running into this same issue, have you checked: https://www.raspberrypi.org/forums/viewtopic.php?t=72587&p=522851

So could it be caused by:

* Incorrect time in the system
* Not having the latest certificates installed

kimmobrunfeldt, 2016-08-31:
I tried both of those suggestions but they didn't help. Also other sites, such as facebook.com give us `ERR_SSL_PROTOCOL_ERROR`.

jareware, 2016-09-04:
This has been addressed in af98f4b9b0fd9736fd36a3aa3ad4f4a26ec8368b, which will be included in the `v1.2` image, which is going into pre-release testing next week.

Thanks for contributing! ❤️

jareware, 2016-09-10:
Released in `v1.2`.

closed 2016-09-10 by jareware