jareware/chilipie-kiosk #10: After reboot, Chrome is replaced with Chilicorn splash screen

laurianttila, 2016-06-29:
Chrome starts up, but is immediately replaced by Chilicorn loader. Fresh setup with nothing done to raspi settings. First boot was ok, but restart broke things

jareware, 2016-07-06:
Whoops, sorry, hadn't added myself to repo watchers and had missed this!

Which image & which raspi version do you have?

laurianttila, 2016-07-06:
I think the problem actually came after I did a apt-get upgrade 😬 . RPi Model 2, Chilipie v 1.0
I did a clean install without any updates and works fine now :).

jareware, 2016-07-06:
Ah, well, I'd maybe recommend against it. :) The goal of the image is to have a known stable config that never needs to be touched once set up. This precludes software updates. Often this won't be as bad as it sounds from e.g. security point of view, since

* the kiosk will often be on a private network, not directly accessible from the internet
* unless you specifically enable e.g. SSH, it doesn't have a lot of attack surface
* and if it were compromised somehow, it's somewhat unlikely it contains any secrets because it's just a web browser

laurianttila, 2016-07-06:
I shall not type upgrade ever again, case closed 😄

closed 2016-07-06 by laurianttila