jareware/chilipie-kiosk #11: Clarify upgrade policy

jareware, 2016-08-24:
Some people have ran e.g. dist-upgrade after installation, which has caused issues. Make sure the README clearly indicates that this is meant to be an appliance.

varac, 2019-08-28:
Damn - I also did an dist-upgrade and now mhy system shows only the splash screen, no chromium anymore.

varac, 2019-08-28:
It would be actually nicer to make chillipie more resilent so you can actually dist-upgrade (I'm talking about upgrading packages, not the distribution), because I want latest security patches appled for raspbian.

jareware, 2019-08-29:
That would indeed be nice!

If you have a PR implementing this, I'd be happy to take a look!

kovacicjanez, 2019-09-23:
Is it possible to upgrade from 2.0.0 to 2.1.0 ? or to 3.0.0-alpha1