jareware/chilipie-kiosk #13: Incorrect resolution

kimmobrunfeldt, 2016-08-31:
Raspberry outputs incorrect resolution to our 1080p monitor.

kimmobrunfeldt, 2016-08-31:
Fix is simply following this guide: http://weblogs.asp.net/bleroy/getting-your-raspberry-pi-to-output-the-right-resolution

Copied here:
> 
> I was setting up a new Raspberry Pi under Raspbian on a Samsung monitor the other day. If you don’t do anything, Raspbian and the Pi will attempt to detect the modes supported by your monitor and will make a choice of what seems best to it. And sometimes it gets that very wrong. In those cases, you’ll need to find what the right mode is and to set it up.
> 
> It took me quite a few attempts before I succeeded, mostly misled by misinformed forum and blog posts. The right post, the one that has all the correct info does exist however: http://www.raspberrypi.org/phpBB3/viewtopic.php?f=26&t=5851. Let me distil this to a short set of instructions, in case you don’t want to dive in and assimilate all that information. Here is what worked for me…
> 
> From the command-line, logged-in as root…
> 
> ### 1. Get the list of what’s supported by your monitor:
> 
> ```
> tvservice -d edid
> edidparser edid
> ```
> 
> ### 2. Find the mode that you want in the resulting list (for me it was “HDMI:EDID DMT mode (82) 1920x1080p @ 60 Hz with pixel clock 148 MHz” with a score of 124,416, which wasn’t the highest, explaining why I had a lower resolution by default). The mode number is the one between parentheses: 82.
> 
> ### 3. Edit the config file:
> 
> ```
> nano /boot/config.txt
> ```
> 
> Find the section about HDMI, uncomment it and set the right group and mode from step 2. If your mode description contains “DMT”, the group should be 2, and if it contains “CEA”, it should be 1, so for me that was:
> ```
> hdmi_group=2
> hdmi_mode=82
> ```
> Exit the editor with CTRL+X, followed by Y.
> 
> ### 4. Reboot:
> ```
> sudo reboot
> ```

closed 2016-08-31 by kimmobrunfeldt

jareware, 2016-08-31:
Wait, why is this closed? :) Should we update `first-boot.md` with this info?

kimmobrunfeldt, 2016-09-01:
@jareware Yeah maybe! I just spat out the information here in case someone searches the issues later.

jareware, 2016-09-04:
http://www.opentechguides.com/how-to/article/raspberry-pi/28/raspi-display-setting.html also contains a good checklist, for future reference.

jareware, 2016-09-04:
This has been addressed in fb41e4a89a08e8044e9b9a5af2e3ef09ce217fb1 and c9a0afd592983124830372ddec903ff54105699b, which will be included in the `v1.2` image, which is going into pre-release testing next week.

Thanks for contributing! ❤️