jareware/chilipie-kiosk #20: The chilipie-kiosk-v1.2.1.img.zip file appears to be corrupted

matthewwicks, 2017-07-31:
when I download the chilipie-kiosk-v1.2.1.img.zip file and try to unzip it is appears to be corrupted. 

```

Extracting to "C:\Users\XXXXX\Documents\"
Use Path: yes   Overlay Files: no
Extracting chilipie-kiosk-v1.2.1.img
Error: Unable to extract "C:\Users\XXXXX\Documents\chilipie-kiosk-v1.2.1.img".

The size of the extracted file (8053063680) does not match the uncompressed size (3758096384) recorded in the Zip file.
Severe Error:  File size doesn't match uncompressed size.

```

matthewwicks, 2017-07-31:
Also check with 7-zip and it reports a CRC error.

jareware, 2017-08-01:
Oh, rats. I was under the impression it decompressed fine, as others definitely have done so... let me check.

jareware, 2017-08-01:
Odd, I just tried downloading the zip file from GitHub, and inflating with:

* The default macOS archive utility
* [The Unarchiver](https://theunarchiver.com/)
* [`unzip` from Brew](http://www.info-zip.org/UnZip.html)

...and all of them finished without reporting errors. 😕 

Can you run the Windows equivalent of:

```
$ md5 chilipie-kiosk-v1.2.1.img.zip
MD5 (chilipie-kiosk-v1.2.1.img.zip) = 5b885b686b5f83d13003f02189e980a5
```

and check that you get the same result?

matthewwicks, 2017-08-01:
I ran the Windows equivalent on the zip file (powershell version):
Get-FileHash -Path .\chilipie-kiosk-v1.2.1.img -Algorithm MD5

Algorithm       Hash                                                                   
---------       ----                                                                   
MD5             BAD950CE0DCAA627C3FCC4932874ABD4

matthewwicks, 2017-08-01:
sorry that was the actually image, here is the zip file data, looks the same to me 

Algorithm       Hash
---------       ----
MD5             5B885B686B5F83D13003F02189E980A5

jareware, 2017-08-01:
Have you tried whether the resulting image works?

matthewwicks, 2017-08-01:
I was using Etcher and it was complaining, but I just tried it again and it seems to copy to the SD card.  But when I plug it into the PI, ti boots up and starts things up but I get lots of FAILED items

EXT4-fs errors  .... bad extra_isize ...
....
Failed to start Braille Device Support
...
etc...

jareware, 2017-08-03:
OK, so you're definitely getting a corrupt image on the SD card. But I'm also getting a different hash on the uncompressed image file:

    MD5 (chilipie-kiosk-v1.2.1.img) = f883e43dce3a0477555fa4444e1853a3

I've uploaded the [uncompressed image](http://s3.jrw.fi/chilipie-kiosk-v1.2.1.img), please try downloading that directly, since it looks like there's something in the ZIP file that prevents it being uncompressed successfully on Windows.

Let me know if that helps.

matthewwicks, 2017-08-03:
That image appears to work, so thanks.  Not sure what is going on with the compressed version.  If you do get a new compressed version that you want me to try let me know.

jareware, 2017-08-04:
OK, I'll try to make sure the "official ZIP" decompresses fine on all platforms for the next release!

closed 2017-08-04 by jareware

jareware, 2017-10-17:
@matthewwicks there's now a [repackaged release of `v1.2.1`](https://github.com/futurice/chilipie-kiosk/releases/tag/v1.2.1-repack) that should work better on Windows.