jareware/chilipie-kiosk #24: Does chilipie-kiosk support GPIO.

faspina, 2018-01-05:
I have a raspbian running chrome in full screen calendar that i hooked a few buttons on to and the python to utlize GPIO to send characters to chromium to control the calendar. Switch views. move forward a month, move back a month. etc.. Can I move this to chilipie-kiosk and make it a little cleaner

jareware, 2018-01-08:
> Does chilipie-kiosk support GPIO

I haven't tried, actually. I have other RasPi's that use GPIO but they're based on Raspbian, whereas chilipie-kiosk is based on Ubuntu. Would be very interested to know! Your setup sounds really cool btw.

Also, the next version of chilipie-kiosk will likely be based on Raspbian as well. Whenever we have a bit of time to work on it again... 😇 

> Can I move this to chilipie-kiosk and make it a little cleaner

You sure can! What kind of upstream changes would you propose to the kiosk?

jareware, 2018-08-17:
As of https://github.com/futurice/chilipie-kiosk/releases/tag/v2.0.0, all the usual GPIO is supported.

closed 2018-08-17 by jareware