jareware/chilipie-kiosk #25: Setting Startup URL via --app does not work

MaXFalstein, 2018-07-01:
Chrome is able to accept `-app "http://github.com"` but it does not work with Chilipie Kiosk.

jareware, 2018-09-19:
Good point, and this is definitely on the radar!

In those cases the biggest hurdle is usually this: most people's dashboards aren't going to be publicly available on the Internet, so you'd still need to authenticate, with a login form or something. Being able to specify that via command line args/SSH is more complicated.

loranger, 2018-09-19:
Yes, you're right. If the webpage uses a http basic auth, it's quite simple to define the user and password in the url, but for a advanced auth process, you have to use a mouse and keyboard…

… Maybe for all the other url, that would be useful ?

jareware, 2018-09-19:
Created #31 to track this.

closed 2018-09-19 by jareware