jareware/chilipie-kiosk #27: dd: invalid number: ‘1m’

olpeh, 2018-08-01:
I think the correct notation is 1M

jareware, 2018-08-01:
From `man dd` on macOS 10.12.6:

> Where sizes are specified, a decimal, octal, or hexadecimal number of bytes is expected.  If the number ends with a ``b'', ``k'', ``m'', ``g'', or ``w'', the number is multiplied by 512, 1024 (1K), 1048576 (1M), 1073741824 (1G) or the number of bytes in an integer, respectively.  Two or more numbers may be separated by an ``x'' to indicate a product.

So I wonder if this is actually platform-specific..?

olpeh, 2018-08-01:
Yes, seems to be platform specific, on Linux Ubuntu 16.04:

>N and BYTES may be followed by the following multiplicative suffixes: c =1, w =2, b =512, kB =1000, K =1024, MB =1000*1000, M =1024*1024, xM =M GB =1000*1000*1000, G =1024*1024*1024, and so on for T, P, E, Z, Y.

jareware, 2018-08-01:
This is why we can't have nice things.

Anyway, like I mentioned on #29, I think we should suggest using https://etcher.io/ in the README anyway!

closed 2018-08-01 by olpeh