jareware/chilipie-kiosk #29: Version 1.2.1 does not seem to fit on a 8gb memory card

olpeh, 2018-08-01:
```
dd: error writing '/dev/mmcblk0p1': No space left on device
7580+0 records in
7579+0 records out
7947157504 bytes (7,9 GB, 7,4 GiB) copied, 585,139 s, 13,6 MB/s
```

[image]

olpeh, 2018-08-01:
If this is the case, the instructions should be updated and the link to Verkkokauppa should be updated as well.

jareware, 2018-08-01:
I'm wondering if #27 might have something to do with this, i.e. being platform dependent for `dd`..? I could have sworn I've also flashed 8 GB cards with that. 🤔 

I'll actually change the README to suggest using https://etcher.io/ for the next release anyway, because it's just so nice to use.

olpeh, 2018-08-01:
Etcher is really nice. And the screenshot is from etcher. 
I'll just go grab a bigger card for now and will close #27 and reject #28

jareware, 2018-08-01:
OK.

Thanks for reporting this, I'll address this with the next build.

Sorry for the trouble.

jareware, 2018-10-23:
This should not be an issue since the 2.x.

closed 2018-10-23 by jareware