jareware/chilipie-kiosk #30: monitor does not wake up

loranger, 2018-09-12:
Hello,

Even if the log says everything works fine

```
Sep 12 07:00:01 chouffe CRON[2627]: (pi) CMD (~/display-on.sh)
Sep 12 07:00:01 chouffe CRON[2631]: (pi) CMD (xdotool key ctrl+R)
Sep 12 07:05:01 chouffe CRON[2682]: (pi) CMD (xdotool key ctrl+Tab)
Sep 12 07:10:01 chouffe CRON[2696]: (pi) CMD (xdotool key ctrl+Tab)
Sep 12 07:15:01 chouffe CRON[2713]: (pi) CMD (xdotool key ctrl+Tab)
Sep 12 07:17:01 chouffe CRON[2724]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 12 07:20:01 chouffe CRON[2744]: (pi) CMD (xdotool key ctrl+Tab)
Sep 12 07:25:01 chouffe CRON[2761]: (pi) CMD (xdotool key ctrl+Tab)
```

My screen (HDMI Monitor does not wake up).

I tried to call `display-on.sh` manually, but it seems the script remains stuck on the `sudo chvt 2` and the screen never went back.

Did I miss something, how can I wake my screen automatically ?

jareware, 2018-09-17:
Sorry to hear you've been having issues.

Are you using a PC monitor or a TV?

loranger, 2018-09-17:
No problem, it's only a tiny issue, everything else works flawlessly ;)

I'm currently using a PC Monitor (BenQ GW 2470-B)

jareware, 2018-09-19:
I wish I had more things to suggest... monitors/TV's sometimes work in mysterious ways. :S

jareware, 2018-10-23:
@loranger any updates?

Jensdiep, 2018-10-23:
I had the same issue with an old monitor fixed like this;
Use crontab -e and reboot at the time you want to wake up the screen.

Works here on a really old monitor 😄

jareware, 2018-10-23:
OK, suppose that counts as a fix. :>

closed 2018-10-23 by jareware

loranger, 2018-10-23:
I went to the same workaround: Rebooting using a cron job ;)

b0bben, 2018-11-28:
I've found a more stable way of doing power off/on for the display-off.sh and display-on.sh script.
My problem (Pi connected to a TV) was that it would always wake up to wrong TTY because of the chtv in the display-on.sh script.

This is what I've used for a couple of day that works great for me:
`sudo vcgencmd display_power 0` for power-off
`sudo vcgencmd display_power 1` for power-on
vcgencmd seems to be (from my quick research) a Broadcom utility for dealing with video chipset on the Raspberry Pi. 

PS:
Tested on Rev.B

kdbqlb, 2020-06-20:
Hi ! Cron doesn t turn off the Screen ? What to do ?