jareware/chilipie-kiosk #31: Pre-setting URL during flash

jareware, 2018-09-19:
We should support writing a file to the `/boot` partition right after flashing, which would contain the default URL the kiosk navigates to after boot.

This would allow fully unattended deployments, meaning you don't need to plug in a keyboard or SSH over, as the kiosk would navigate to the desired URL on its own, after plugging in the power.

Bonus points for being able to provide e.g. HTTP basic auth in that file.

bluepackets, 2018-11-15:
Thanks for your work on this project :) Yes, agreed. I had also tried "FullPageOS" which had a file in /boot/fullpageos.txt which allowed us to set the URL during the flash rather than having to use a keyboard to set it up.

The other issues is that SSH appeared to be disabled by default and also the keyboard was set to a different locale (I'm in Australia) so things such as a colon ( : ) would do a different character. All easily overcome, but could be added to the readme I suppose :)

jareware, 2018-11-19:
Thanks for the feedback!

The instructions for changing the keyboard layout (among other things) is in the [first boot document](https://github.com/futurice/chilipie-kiosk/blob/master/docs/first-boot.md#system-configuration), but you're not the first person to not find it immediately... where do you think it'd be more prominent? On the main README..?

bluepackets, 2018-11-19:
Ahhh, so it is. Apologies :)

Perhaps instead of just "Change your keyboard layout" it says "Change your keyboard layout (it is configured by default to X)" ?

jareware, 2018-11-20:
That could be good, yeah.

Also, before open-sourcing this project, most users were internal, and a Finnish default keyboard layout was fine. Now that the user base is more global, it might be a better idea to default to the US one... 🤔

janka, 2019-06-10:
Setting an URL is easily done in /home/pi/.xsession:

```
#chromium-browser --start-fullscreen --window-size=1920,1080 --disable-infobars &
chromium-browser --disable-infobars --kiosk http://somewhere.com &
```

Should also be possible to add basic auth:

`https://username:password@somewhere.com`

crisp2u, 2019-09-13:
Could I drop another idea here ? How about using Chrome Debugging Protocol for this ? I'm planing o build a prototype with chilipie and I'm planning to try https://github.com/cyrus-and/chrome-remote-interface. I'll keep you guys updated