jareware/chilipie-kiosk #33: Prevent closing or force-restart of chromium

Levvy055, 2018-10-01:
There is a page(not mine) I am displaying with some buttons.
Each button is opening popup in fullscreen mode. There are no buttons to close it so.
I only can press Alt+F4 to close it. Is there some possibility to make X button of windows visible?
Because people sometimes press Alt+F4 too many times and closes main window which I want to prevent because after that I need to switch to console and start chromium process again or restart Pi.

jareware, 2018-10-02:
The kiosk uses the [Matchbox window manager](https://en.wikipedia.org/wiki/Matchbox_(window_manager)), which is meant for these "embedded" use cases, so there's no window controls by default.

Some avenues you may try:

* Not running Chromium in full screen mode, which allows you to just use the X button of the tab/window to close it
* Writing a [Tampermonkey](https://chrome.google.com/webstore/detail/tampermonkey/dhdgffkkebhmkfjojejmpbldmpobfkfo?hl=en) userscript that adds a close button to the page itself
* Trying something like `matchbox-keyboard` (probably not going to be useful here, though)

Hope this helps.

Levvy055, 2018-10-02:
I think the second option will be the best because in first it's opening a new window so no frame with X button.

jareware, 2018-10-02:
Sounds good.

But also somewhat outside the scope of this project, so I'm going to go ahead and close this one. :)

closed 2018-10-02 by jareware