jareware/chilipie-kiosk #34: How to change keyboard settings

FreyaFreed, 2018-10-18:
Is there a way to change keyboard settings? It can be quite frustrating for those of us who don't have Finnish/Swedish keyboards

jareware, 2018-10-19:
Hey @FreyaFreed! Indeed the keyboard layout should have a sensible default... The trouble is picking which one. Based on the user base it's been Finnish thus far, but maybe a US layout would be more neutral..?

But I digress; the [First Boot document](https://github.com/futurice/chilipie-kiosk/blob/master/docs/first-boot.md#system-configuration) has instructions on how to change the keyboard layout, and other things on your kiosk.

Hope that helps!

jareware, 2018-10-23:
I'll go ahead and assume this got sorted out. :)

closed 2018-10-23 by jareware