jareware/chilipie-kiosk #36: Alternatives to wifi conection (BerryLan)

matristain, 2018-11-08:
This is an alternative way to get rasperry connected to WLAN more easily and keyboard less

[BerryLan](https://berrylan.app)

This is the installation instructions for a raspian
sudo apt-get install dirmngr
echo "deb http://repository.nymea.io stretch main" | sudo tee /etc/apt/sources.list.d/nymea.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
sudo apt-get update
sudo apt-get install nymea-networkmanager
sudo systemctl disable dhcpcd
reboot

But could be nice if this can me included and informed on the  first-boot document (offline version)

jareware, 2018-11-09:
Wow, really neat utility, wasn't aware of this!

Given that it adds a dependency on an iOS/Android app, I'm not sure I want to ask that of users, though.

Also, maybe I'm missing something obvious, but how does the Raspberry Pi know which WiFi to join? What if the WiFi is password protected? 🤔

matristain, 2018-11-09:
The app connects to the Raspberry thru Bluetooth,
In the app you'll see the list of available WiFi , when you select one It ask you for the password,
This is send to the Raspberry and use this info to connect to that WiFi.

I Already try it and is very useful.

The iOS/Android app dependency is not the best , but the is better than getting a keybord.

jareware, 2018-11-12:
Very interesting!

I'll keep this open until I we have some time to evaluate incorporating this!