jareware/chilipie-kiosk #37: Refreshing via crontab does not work

senseitsa, 2018-11-22:
My RPi is showing a website of local IP, and refreshes every 30 seconds via jquery. When the network goes down, the RPi brings up the "Page not found" page.

This means that the page cannot reload when the network comes up again, as it is stuck on the Page Not Found page.

I have tried reloading the page every minute using crontab (* * * * * xdotool key ctrl+R), however this doesn't seem to work. Any suggestions on how I can reload the page using crontab or a shell script?

jareware, 2018-11-22:
That sounds odd.

I just tried disconnecting from my network so I got the "page not found" page on Chrome, turned the network back on, and sent the reload keypress, and it loaded just fine.

Can you recover from that situation in any other way? For example, using the keyboard/mouse to reload the page? Does that work?

matristain, 2018-11-22:
In my case y have to do a 
export DISPLAY=:0
Before any xdotool key , to make it work properly.
not sure if its your case or how to integrate it on crontab , but hope this help.

jareware, 2018-11-22:
https://github.com/futurice/chilipie-kiosk/blob/6a4a3e6f8aac958c17059f99c2991eaa496c0368/home/crontab.example#L1

It already should be set for all commands in the crontab, though... 🤔

senseitsa, 2018-11-23:
Thanks for all your help. I've tried the following:
```
$ xdotool key ctrl+R
Error: Can't open display: (null)
Failed creating new xdo instance
```
Then tried the export command first, which worked fine:
```
$ export DISPLAY=:0
$ xdotool key ctrl+R
$ 
```

So I've updated the crontab entry to the following:
`* * * * * export DISPLAY=:0;xdotool key ctrl+R`

closed 2018-11-23 by senseitsa

jareware, 2018-11-23:
Hmm, very curious!

If it's not too much to ask, could you still try and see if you can make it work by changing the `DISPLAY=:0.0` **at the top of the crontab file** to `DISPLAY=:0` to see if that fixes it as well?

Because setting it at the top should definitely set it for the environment of all cronjobs.

jareware, 2018-11-25:
According to [this answer](https://stackoverflow.com/a/10657111) it should work on Debian, but should take a closer look.

connoleg, 2019-10-28:
I'm struggling with this crontab doesnt change tabs. On boot the first tab displays but never changes. I can manually change the tab with export DISPLAY=:0 and xdtool ctrl-Tab etc.