jareware/chilipie-kiosk #38: Remote view/management for browser and content

rradar, 2018-11-22:
Hello,

this is a very nice kiosk setup! I used to have raspbian with chromium kiosk with some unclutter and a few bits more. But this one is far more advanced with **nightly reboot, hourly refresh, zoom, password saving** etc. :+1: :+1: :+1: 

I'm only **missing one feature which is to see (and change) the browser (and content) remotely**. My use case is that the screen is few hundred kilometers away and I need to check if it has a proper content loaded. Also I could get a call that something is not properly shown and I need to be able to see what is actually shown to troubleshoot it properly.

My setup right now is using Teamviewer for this. I tried to install it via command line in chilipie-kiosk and actually it was successful (installing the teamviewer-host_armhf.deb with around 100MB of dependencies). I even could set it up properly via command line (running 'teamviewer setup' command). Problem is that I can only connect when the console is active on tty2. Otherwise I can't connect.

I'm not keen on using Teamviewer at all but **just need ANY remote management solution to be able to see and change the content of the browser**. For sure **FOSS** is preferred!

And thank's a lot for this great project!
Greetings

jareware, 2018-11-22:
Thanks for the feedback, I'm glad it's working out for you!

Personally I haven't used TeamViewer on Linux, and I have no idea how to make your setup work. That said, it definitely *sounds* like it's just a small configuration issue somewhere, and someone more knowledgeable on the product might be able to figure it out!

VNC is often used for remote desktop connections on Linux; a quick google will give you tons of options in terms of both server and client. You can (and should!) also run it securely over SSH, which the kiosk already has.

Hope this helps!

matristain, 2018-11-22:
> Hello,
> 
> this is a very nice kiosk setup! I used to have raspbian with chromium kiosk with some unclutter and a few bits more. But this one is far more advanced with **nightly reboot, hourly refresh, zoom, password saving** etc. 👍 👍 👍
> 
> I'm only **missing one feature which is to see (and change) the browser (and content) remotely**. My use case is that the screen is few hundred kilometers away and I need to check if it has a proper content loaded. Also I could get a call that something is not properly shown and I need to be able to see what is actually shown to troubleshoot it properly.
> 
> My setup right now is using Teamviewer for this. I tried to install it via command line in chilipie-kiosk and actually it was successful (installing the teamviewer-host_armhf.deb with around 100MB of dependencies). I even could set it up properly via command line (running 'teamviewer setup' command). Problem is that I can only connect when the console is active on tty2. Otherwise I can't connect.
> 
> I'm not keen on using Teamviewer at all but **just need ANY remote management solution to be able to see and change the content of the browser**. For sure **FOSS** is preferred!
> 
> And thank's a lot for this great project!
> Greetings

rradar, can you share your configuration for teamviewer , I tryed my self but i couldn't log in to my account at "teamviewer setup" (always get connection error) , y personally like this alternative to other because of the "Management console" to keep connected and monitoring behind  router without port forwarding.

Now I am using remot3.it (USD $2 / Device) to keep ssh connection to 10 devices (for now). 

is there a better way to keep connected to remote devices behind router?

Best regard , thank you in advance.

rradar, 2018-11-27:
@matristain wrote:
> rradar, can you share your configuration for teamviewer , I tryed my self but i couldn't log in to my account at "teamviewer setup" (always get connection error) , y personally like this alternative [...]

I did run (and installed) teamviewer all locally on tty2 with a attached keyboard on the device while it is plugged with a Ethernet cable connected to world wide web. The setup worked directly on the console. Just need to take sure have two-factor (otp) enabled or otherwise you need to click a link you get via mail in time (I think the timeout is quite low... maybe 120sec?).

But problem is that teamviewer is not able to connect at all to the device while the kiosk is running and showing content. Just when you a locally showing the console (tty2) it will be able to connect to the device with teamviewer (which is quite useless).

I'm right now searching for possibilities to **remote manage** tty1 which is running on **matchbox**. Let's see if I get something usable!

rradar, 2018-11-27:
Good news everyone!

I'm able to connect with **vnc** (using x11vnc on the chilipie-kiosk) to the browser and also control it remotely!

It was as simple as:

`sudo apt install x11vnc`

and running this command in command line:

`x11vnc -usepw -forever -display :0`

When switching back to the browser I'm able to connect via vnc from a client in the same network. I can use another client in the same network with teamviewer to hop to it from everywhere in the world.

Last thing I need to figure out how to auto start it.. _but this should be easy!_

Maybe this (vnc preinstalled/preconfigured) could be included in the next awesome chilipie release! :+1:

rradar, 2018-11-27:
And here is the missing piece to auto start the vnc server:

`sudo nano /lib/systemd/system/x11vnc.service`

```
[Unit]
Description=Start X11VNC
After=multi-user.target

[Service]
Type=simple
ExecStart=/usr/bin/x11vnc -display :0 -auth guess -forever -loop -noxdamage -repeat -rfbauth /etc/x11vnc.pass -rfbport 5900 -shared

[Install]
WantedBy=multi-user.target
```

`sudo systemctl enable x11vnc.service`

reboot & profit! :+1:

jareware, 2018-11-28:
@rradar thanks a lot for detailing your setup! 🎉 

Indeed, I think this could make a great addition to the next version, either in documentation, or built-in by default.

matristain, 2018-11-29:
@rradar thak you , I'll try it tonight.

In addition next step is to create a revers ssh tunnel to a AWS server with fixed IP from the raspberry.

Something like "ssh -R 5901:localhost:5900 sourceuser@FIXED_IP"

To use VNC client connect to: FIXED_IP Port 5901 and access the raspberry (behind a router).

Hope I can make this work I'm not that into Linux/command/tunneling But teoricaly that should work.

For now I use a remot3.it service to access raspberrys behind a router (only ssh) , but is a paid service.

I need an alternative free of paid, you have any other option?

isabellaalstrom, 2019-04-28:
This would still be a great idea.

jareware, 2019-04-29:
I agree! Since there's been interest for this feature here and elsewhere, would any of you want to weigh in on what would be the best and most versatile solution for remote management:

1. Persistent SSH port forward with e.g. `autossh`. Pro: Allows full GUI management over VNC, meaning you can open/close tabs, interact with pages, etc. Con: Requires a jump-host running SSH, which is always on.
1. Polling a URL over HTTP, and navigating the kiosk to where the server tells it to. Pro: Really simple, you can use e.g. Amazon S3 to host this so it's always up and basically free. Con: Only allows simple navigation (or reload) commands, can't interact with pages, extensions etc.

Thoughts?

isabellaalstrom, 2019-04-29:
Absolutely the first option.

c0un7-z3r0, 2019-05-06:
First option as well

wujood, 2020-07-24:
I just had the same issue and searched for a solution. I have not found one, but i found a hint for the solution. However i am not very familiar with this stuff. Here is what i found: https://community.teamviewer.com/t5/Linux/Unable-to-connect-to-TeamViewer-headless-after-xsession-started/td-p/10608

It seems that lightdm and openbox solved the issue. Does anyone understand that solution and maybe can describe how to get to that solution?

phrfpeixoto, 2022-09-30:
> And here is the missing piece to auto start the vnc server:
>
> `sudo nano /lib/systemd/system/x11vnc.service`
>
> ```
> [Unit]
> Description=Start X11VNC
> After=multi-user.target
>
> [Service]
> Type=simple [...]

I tried your file, and the system gives this out on the syslog:
```
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 passing arg to libvncserver: -rfbauth
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 passing arg to libvncserver: /etc/x11vnc.pass
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 passing arg to libvncserver: -rfbport
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 passing arg to libvncserver: 5900
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 x11vnc version: 0.9.16 lastmod: 2019-01-05  pid: 6271
Sep 30 15:51:08 chilipie-kiosk x11vnc[6316]: xauth:  unable to generate an authority file name
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 -auth guess: using 'XAUTHORITY=/tmp/serverauth.4uL3birRPE' for disp=':0'
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 Using X display :0
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 rootwin: 0x27d reswin: 0x1600001 dpy: 0x9318f8
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 ------------------ USEFUL INFORMATION ------------------
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 Wireframing: -wireframe mode is in effect for window moves.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   If this yields undesired behavior (poor response, painting
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   errors, etc) it may be disabled:
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08    - use '-nowf' to disable wireframing completely.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08    - use '-nowcr' to disable the Copy Rectangle after the
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08      moved window is released in the new position.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   Also see the -help entry for tuning parameters.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   You can press 3 Alt_L's (Left "Alt" key) in a row to
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   repaint the screen, also see the -fixscreen option for
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   periodic repaints.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 XFIXES available on display, resetting cursor mode
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   to: '-cursor most'.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   to disable this behavior use: '-cursor arrow'
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   or '-noxfixes'.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 using XFIXES for cursor drawing.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 GrabServer control via XTEST.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 Scroll Detection: -scrollcopyrect mode is in effect to
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   use RECORD extension to try to detect scrolling windows
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   (induced by either user keystroke or mouse input).
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   If this yields undesired behavior (poor response, painting
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   errors, etc) it may be disabled via: '-noscr'
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   Also see the -help entry for tuning parameters.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   You can press 3 Alt_L's (Left "Alt" key) in a row to
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   repaint the screen, also see the -fixscreen option for
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   periodic repaints.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 XKEYBOARD: number of keysyms per keycode 7 is greater
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   than 4 and 51 keysyms are mapped above 4.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   Automatically switching to -xkb mode.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   If this makes the key mapping worse you can
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   disable it with the "-noxkb" option.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08   Also, remember "-remap DEAD" for accenting characters.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 X FBPM extension not supported.
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]: 30/09/2022 15:51:08 X display is capable of DPMS.
[... cut ...]
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]:   Minor opcode of failed request:  1 (X_ShmAttach)
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]:   Serial number of failed request:  56
Sep 30 15:51:08 chilipie-kiosk x11vnc[6271]:   Current serial number in output stream:  118
Sep 30 15:51:09 chilipie-kiosk x11vnc[4675]:  --- x11vnc loop: sleeping 2000 ms ---
Sep 30 15:51:11 chilipie-kiosk x11vnc[4675]:  --- x11vnc loop: 12 ---
Sep 30 15:51:11 chilipie-kiosk x11vnc[4675]:  --- x11vnc loop: waiting for: 6428
```