jareware/chilipie-kiosk #41: Rotate the screen and Touchscreen

Fyb3roptik, 2019-01-23:
I can rotate the screen 90 degress by doing display_rotate=1, but the touch does not do this. Every tutorial I have seen has more text files than what is in here by default. Does anyone know how to make the screen and touchscreen rotate 90 degrees?

RPi 3 with 7" touchscreen

GendoRokubungi, 2019-02-09:
If this is a cheap Chinese screen, you'll need to install the drivers for your screen and then edit boot/config.txt to add `display_rotate=1` as per the instructions in first-boot.md. Check under /usr/share/X11/xorg.conf.d/ for a libinput conf file. You can add

```Option "CalibrationMatrix" "0 1 0 -1 0 1 0 0 1"```

in the `touchscreen catchall` section. Check out the PDF manual for this screen for the general idea - https://www.waveshare.com/wiki/5inch_HDMI_LCD_(H)

jareware, 2019-03-01:
Great pointers, thank you @GendoRokubungi!

@Fyb3roptik, did that help you get it sorted out?

closed 2019-04-04 by jareware

lbillett, 2020-04-19:
I had some trouble with this today (am bad in linux, but I get by). I was  _delighted_ to find my Viewsonic TD2421 worked perfectly, touch screen and all, right out of the box on Chilipie 2.1.0.  What a great repo for those looking for a lean web kiosk with no BS! So awesome. 

Setting `display_rotate=3` in `/boot/config.txt` worked great for the display, but (as expected) the touch screen input was now incorrect. 

`DISPLAY=:0 xinput_calibrator` had no effect (even after finding out I had to install it) not sure why. It would take time to run and finish, but it never asked me for any inputs or anything.  

There were some `xinput set-prop` solutions out there that were more involved, none of which I managed to get working. 

Using  `Option "CalibrationMatrix" "0 1 0 -1 0 1 0 0 1"` in my `/usr/share/X11/xorg.conf/40-libinput.conf` was _close_ but the display went all whacky. The lower half of the screen stopped working (though the pointer could be seen on it), the width seemed to revert to normal wide-screen after any inputs were received from the screen or keyboard, spilling off of the viewable area of the monitor, and my Pi3 started to overheat! I tried lots of combinations of these and even found an [awesome tool](https://codepen.io/GottZ/full/d73f2f844b52b91b7457febce2d1b18c/) to help visualize the transformations, but I never made it home. 

What ultimately worked for me was a combination of installing the evdev driver (whatever that is) with:
`sudo apt-get install xserver-xorg-input-evdev`

Then hacking up my `/usr/share/X11/xorg.conf/40-libinput.conf` and changing the `Driver "libinput"` to `Driver "evdev"` as [done in this thread](https://www.raspberrypi.org/forums/viewtopic.php?t=179477), allowing use of the SwapAxes and InvertX and InvertY options like [outlined here](https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=230554&p=1413454#p1413454). My resulting section looked as:

`Section "InputClass"`
`  Identifier "libinput touchscreen catchall"`
`  MatchIsTouchscreen "on"`
`  MatchDevicePath "/dev/input/event*"`
`  Option "SwapAxes" "1"`
`  Option "InvertX" "true"`
`  Option "InvertY" "true"` (if you have `display_rotate=1` I think)
`  Driver "evdev"`
`EndSection`

Is probably totally backwards way of doing it, but it worked for my screen perfectly!

Kristories, 2022-07-27:
Thanks @lbillett

It work for RPi3. 

`/boot/config.txt` : 

```
#dtoverlay=vc4-fkms-v3d
dsplay_rotate=3
```

`/usr/share/X11/xorg.conf/40-libinput.conf` :
```
Section "InputClass"
  Identifier "libinput touchscreen catchall"
  MatchIsTouchscreen "on"
  MatchDevicePath "/dev/input/event*"
  Option "SwapAxes" "1"
  Option "InvertX" "true"
  Driver "evdev"
EndSection
```

commit cb17c26e (2019-04-04): Add notes about rotating more exotic screens

Thanks @GendoRokubungi!

Closes #41