jareware/chilipie-kiosk #42: Documentation on what packages were used

NoFear13X, 2019-01-25:
I'm somewhat new to the Linux world, and I absolutely love this build, but I need a faster device than a Pi for the website/video we're running in a loop. Can I somehow run this on a different platform than a Pi, and if not, is there any documentation of how exactly this build was created? I'd like to try to duplicate the majority of it on a fresh build of Ubuntu Server, but if I can avoid all of that redundant work, then even better.  Thanks for an awesome project!

jareware, 2019-03-01:
Hey, sorry for taking some time to get back to this, I've been away from the keyboard for a while.

You're not the first one to ask for something like this! Though I haven't (yet) heard back from anyone having actually done it.

The preparation of the image is fully automated, and documented as a script [here](docs/image-setup.sh). The setup on top of regular-old Debian would probably be quite similar, with some caveats, such as:

1. You're not creating a disk image on an SD card
1. You're (probably) not on an ARM architecture
1. You won't have the convenience of `raspi-config` for some system settings

That said, none of those issues are hard to overcome, just take a bit of work.

If you make progress toward such a non-RasPi build, please check back!

NoFear13X, 2019-03-01:
> script here <http://docs/image-setup.sh>. The setup on top of regular-old
>    2. You're (probably) not on an ARM architecture
>    3. You won't have the convenience of raspi-config for some system

Awesome, thank you!!

closed 2019-03-03 by jareware