jareware/chilipie-kiosk #46: Calibrate touchscreen

JamesCullum, 2019-02-19:
Thank you for the easy to use package!

I am using a 5" touchscreen, which requires calibration to be precise.
For this, I need to install `apt-get install xinput-calibrator` (which works) and then go into the GUI to Menu-->Preferences-->Calibrate Touchscreen. 

How do I get to the GUI where I could do this? I've noticed that `startx` is installed, but I assume focuses on Chrome.

Do you know how to get to this menu / get to the GUI in this image?

jareware, 2019-03-01:
Hey, sorry for taking some time to get back to this, I've been away from the keyboard for a while.

I don't have a test device at hand right now, but have you tried just getting a terminal with `Ctrl + Alt + F3`, and running something like `sudo DISPLAY=:0.0 xinput_calibrator`?

JamesCullum, 2019-03-02:
Hey @jareware, thank you for the response!

If I run your command, I get 

> No protocol defined
> Unable to connect to X server

If I change the ID to 1.0 or 2.0 I don't get the `No protocol defined`, but the error connecting to X stays.

jareware, 2019-03-03:
Uhh, interesting. Like I said, I don't have such a device at hand, and that makes testing quite hard. Wish I could be of more help.

Let us know if you figure it out!

Kimmax, 2019-03-28:
Running _xinput_calibrator_ using `DISPLAY=:0.0 xinput_calibrator` works. It's important to run this as a normal user, not root.

jareware, 2019-04-01:
Great to hear!

By the way, are you guys talking about the official RasPi touchscreen module, or something else..?

Kimmax, 2019-04-02:
I'm using a 10.1" 1280x800 "IPS LCD" (?!) Touch display from SunFounder. You can find it [here](https://www.amazon.de/Raspberry-10-1-Display-Touchscreen-SunFounder/dp/B0776VNW9C).  
It's a really cool display for all sorts of projects and looks modern and clean

jareware, 2019-04-04:
Added your tip to `first-boot.md`, thanks @Kimmax!

closed 2019-04-04 by jareware

PoloB12, 2020-02-10:
xinput_calibrator is not included anymore in the image  v2.1.0  ?

Kimmax, 2020-02-10:
@PoloB12 _xinput_calibrator_ was never included, install using `apt-get install xinput-calibrator`

PoloB12, 2020-03-09:
I assumed it was installed as was mentioned in `first-boot.md` ; had installed manually already, thanks.

Had some issues calibrating  a rotated screen though, I found this link very helpful solving that :

[https://gist.github.com/KurtJacobson/37288a0300a9c1b3e859c8dcff403300](https://gist.github.com/KurtJacobson/37288a0300a9c1b3e859c8dcff403300)

commit 53788bb6 (2019-04-04): Add instructions on xinput_calibrator to first-boot.md

Thanks @Kimmax!

Closes #46