jareware/chilipie-kiosk #47: Green screen on chromium-browser with video at fullscreen

matristain, 2019-03-11:
Is not really a direct chilipie-kiosk issue but i'm not sure how can I solve this in a remote device.

Not sure if some one ever heard about it , there is a "common" issue in chrome if you play video at full screen it shows a green screen, in my case the script also beak as this is a remote device I don't have many more info just that it happens every single time it plays a video.

Its safe to update chromium  or reinstall it at all?  and how? :P 

or any other solution for this issue.

*I have about 10 devices an only 1 has this issue.

jareware, 2019-03-12:
I would agree that it doesn't sound specific to the kiosk... so I would rather focus on researching whether other people are having the same issue with Chromium on Raspbian, and/or specific Pi hardware (as it sounds like it's a HW acceleration issue).

jareware, 2019-04-04:
Did you end up finding a solution @matristain?

matristain, 2019-08-22:
I've tryed meny thins on this devives but what I think was the solution is this:
I've add this lines to the .xsession file before the chomium-browse line:

sudo rm -r ~/.cache/chromium
sudo rm -r ~/.config/chromium/Default/

MAC=$( cat /sys/class/net/eth0/address | tr --delete : )

chromium-browser --start-fullscreen --autoplay-policy=no-user-gesture-required --window-size=1920,1080 --disable-infobars https://mydomain.com/device/$MAC &

So far I don't have this issue any more.

PS: I don't know if this interfere with the "open last page" as I use the MAC Address to redirect each device wherever i want from my server.