jareware/chilipie-kiosk #49: chromium needs rng-tools

mrvanes, 2019-03-18:
Without rng-tools Chromium refuses to start due to a starved random pool entropy (on pi zero, at least).

jareware, 2019-03-21:
Interesting, haven't seen this happen!

What do you mean by a "headless" Pi in this instance?

mrvanes, 2019-03-21:
I actually mean without keyboard or mouse, only touchscreen.

jareware, 2019-04-04:
OK!

Does it then work if not running headless?

Also, you made me very curious: what's your use case for running `chilipie-kiosk` headless, and on a Zero no less? :)

mrvanes, 2019-04-04:
I don't know, I never attached a keyboard or mouse. I use it to display HomeAssistant Lovelace webinterface next to my thermostat and use the built-in touchscreen to toggle scenes ;)
The Pi0 is a bit underpowered I admit, but once loaded it reacts fairly well to the touchscreen.

jareware, 2019-04-04:
Very cool use case!

I'll keep this issue open so I can try to bundle the fix (just `apt-get install rng-tools` worked for you, right?) for the next release, and try it on a Zero.

mrvanes, 2019-04-04:
Yes, ```apt install rng-tools``` was the solution for getting chromium to start (apt-get is being deprecated).

jareware, 2019-04-10:
We'll get a regular version-bump release out the door first, and then make a new one with some new features.

Sit tight!

ungarida, 2019-05-23:
Hi Jarno, thank you for this amazing project, really handy! I have a similar use case. I want to use pi zero because it will be used to show monitoring dashboard on a big screen. Is this still happening?

jareware, 2019-06-05:
I haven't gotten my hands on a Zero yet for testing, so can't confirm or deny, unfortunately!

Also, while the Zero is seriously cool, I've seen some people get a bit unhappy with how janky it gets when rendering a heavier dashboard. Chromium has to do a lot of work for those, and the Zero can be a bit underpowered for that.

Finally, if someone can confirm that it's just this package that's the fix on a Zero, would be much appreciated!

mrvanes, 2019-06-06:
As the reporter I surely can confirm that installing rng-tools resolved my problem.

commit afa230e4 (2020-03-13): Install rng-tools to support Pi Zero.

Closes #49
closed 2021-12-28 by jareware