jareware/chilipie-kiosk #51: Watchdog

matristain, 2019-04-09:
For this Raspian / kernel version to activate watchdog
Simply edit /etc/systemd/system.conf 
to include 
"RuntimeWatchdogSec=14"?

jareware, 2019-04-10:
That's a good point! According to [this answer](https://raspberrypi.stackexchange.com/questions/85193/rpi3-is-the-hardware-watchdog-enabled-by-default) it's enabled by default, but I'll need to check.

jareware, 2019-04-10:
I can also see the watchdog being initialized in `dmesg`.

Do you have a way of testing whether it actually works? I'm not sure I do... 🤔

jareware, 2019-04-24:
Based on my best understanding, the watchdog IS active, (at least) as of release `v2.1.0`.

Closing this for now, please reopen if you have more info.

closed 2019-04-24 by jareware