jareware/chilipie-kiosk #56: Is it possible to make boot time faster?

janka, 2019-06-10:
I have not made any comparisons, but my impression is that boot is fairly slow, around 2½-3 minutes before browser is displaying the page.

It is not that big of a deal, just wondering if there where some low hanging fruits somewhere...

jareware, 2019-06-12:
It definitely could be faster.

Nothing obvious comes to mind as to where trim any fat, but if you have any ideas I'd be more than happy to try and incorporate them.

Also keep in mind that your choice of hardware & website matters. Booting up & opening google.com on a Pi 3 will be a LOT faster than booting up & opening `$someBloatedSinglePageApp` on a Pi Zero.

jareware, 2019-08-21:
I'll go ahead and close this, until we have some concrete ideas for improvement.

Feel free to reopen when we do!

closed 2019-08-21 by jareware