jareware/chilipie-kiosk #58: Rotate

danielwje, 2019-06-11:
We're using Hanspree HT225 touch screens with this. We've added the Rotate 90 degree command as we wish to use in Portrait but the touch is all off when in Portrait mode. Am I missing something?

jareware, 2019-06-12:
Did you look at #41? Does any of that get you where you're going?

danielwje, 2019-06-12:
Yep but I am not 100% sure how to do this - can you help with some instructions?

Check under /usr/share/X11/xorg.conf.d/ for a libinput conf file. You can add

Option "CalibrationMatrix" "0 1 0 -1 0 1 0 0 1"

jareware, 2019-06-13:
Sorry, don't have a similar display to fiddle with, and never have!

danielwje, 2019-06-15:
I solved it by following https://www.waveshare.com/w/upload/8/82/5inch_HDMI_LCD_%28H%29_User_Manual_EN.pdf

jareware, 2019-06-17:
Thanks for reporting back! I added this to the first-boot doc (29ee021f08758568d6bbd0ad568ccc0327353b00) so anyone else stumbling on the same issue might have an easier time!

closed 2019-06-17 by jareware