jareware/chilipie-kiosk #59: Black screen on some devices.

matristain, 2019-06-21:
I'm not sure if this is an actual Chilipie-kiosk issue or a setup issue , but hope some one can help me solve this issue.

I have about 10 devices with the same setup (Chilipie with remote.it service and VNC, auto reboot at 5am), 
I also install scrot to generate "screenshot" images from what is showing the raspberry generating a screenshot every 10 minutes end updating via curl to my server to keep an eye on that each one is displaying.

So the issue is 2 of this devices show black screen on every screenshot also when VNC all I see is a black back ground and nothing else (VNC is OK I can see the mouse pointer) ssh device also is OK and shows chome-browser activity as it is working properly.

The only way to solve this for now is reboot.

Maybe all explanation is no needed , but also wanted to share if its useful fore some one.

thank you in advance.

jareware, 2019-08-21:
Hmm, sounds odd.

Do the Pi's still show the browser on their HDMI, even if the screenshots are blank?

It definitely seems like a built-in solution for #38 would make a lot of folks' life easier, including yours.

jareware, 2020-03-13:
This issue has been had to reproduce, and we're hoping it goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware