jareware/chilipie-kiosk #60: Auto-Reconnect to wifi

akamarkman, 2019-06-28:
When there's no wifi signal present during boot-up you're out of luck until a restart.

Would it be possible to add a feature toggle in settings to make it automatically check for an IP every 15 minutes or so and attempt to reconnect with the stored credentials?

Something like what this script does: [https://gist.github.com/mharizanov/5325450](https://gist.github.com/mharizanov/5325450)

jareware, 2019-06-30:
That definitely sounds like good feature!

If you can adapt the linked script to work on the kiosk, a PR would be very welcome!

akamarkman, 2019-07-03:
Cool, so quick question first: Is there a reason why this feature was removed? Or perhaps more accurately, _was_ it removed? 

The reason why I ask is because my colleague tells me their pi automatically reconnects to wifi but I haven't confirmed or compared OS version or wifi utilities.

jareware, 2019-07-03:
> Or perhaps more accurately, was it removed?

The only thing I can say is it definitely wasn't removed, at least knowingly!

The distro is very closely based on Raspbian, and if you look [here](https://github.com/futurice/chilipie-kiosk/blob/master/docs/image-setup.sh) you can see everything that's done on top of a vanilla Raspbian. Nothing AFAICT has to do with what you're describing. 🤷‍♂ 

Have you tried flashing a vanilla Raspbian and seeing if it exhibits the same behaviour?

jareware, 2020-03-13:
This issue has been had to reproduce, and we're hoping it goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware