jareware/chilipie-kiosk #61: Any timeline for building an image with Raspbian Buster to support the Pi4?

brettpappas, 2019-07-01:
Wanted to see if there is any work or plans to create a new image based off Raspbian Buster Lite in order to add support for the new Pi 4's. 

Could useful to provide a provisioning script if someone wanted to build their own image also (using something like ansible).

jareware, 2019-07-01:
Yes, we've been meaning to order one to the office, and then update the image. Thanks for reminding me...

The image preparation is already [scripted to a very large degree](https://github.com/futurice/chilipie-kiosk/blob/master/docs/image-setup.sh). Should be simple enough to modify for building other images, if one is so inclined!

jareware, 2019-07-02:
A lot of shops are currently saying something along the lines of:

> Please note that delivery dates for Raspberry Pi 4 models are uncertain. We estimate that the 4 GB model will be delivered in September. This is an estimate and can be changed to an earlier or a later date with short notice.

So this may need to wait a bit, for practical reasons.

brettpappas, 2019-07-02:
Thanks - I will take a look at the prep script! I have a feeling that the 4GB version will be hard to get and have a high demand for the rest of 2019.

benderllin, 2019-07-20:
It would be fantastic to be able to use it with the PI 4. Congratulations your project is simple but really useful, thank you!

jareware, 2019-07-20:
Thanks for the feedback @benderllin, appreciated!

Still waiting to get some hands-on time with the 4, but should happen within a few weeks now!

As mentioned above, the image setup is mostly scripted, so if you have one and have a bit of extra time, you can probably already make your own quite easily!

darshana-danrox, 2019-07-26:
Hi Guys 

if you guys can get me pi  4 I would like to start this  project  .
any how I'll start source build from my end and then you guys test it and 

Cheers 
danrox

benderllin, 2019-07-26:
I can send you one 2 GB version, i contact you. Thanks

jareware, 2019-07-26:
I also got my hands on a 4, I should have time next week to start fiddling with them :tada:

Numiz, 2019-08-13:
Is there any update to this? I've paused all my projects in anticipation of this :)

benderllin, 2019-08-15:
> 
> 
> Is there any update to this? I've paused all my projects in anticipation of this :)

here working on it, soon we report

jareware, 2019-08-15:
@benderllin great to hear, thanks a lot!

jareware, 2019-08-18:
Took a first look at this: it seems that while nothing on the software package side has changed too much, some of the auto-login stuff has, and e.g. the Plymouth splash screen isn't working anymore.

jareware, 2019-08-18:
Seems that the old method of just [`systemctl enable getty@tty2`](https://stackoverflow.com/a/42742212) doesn't do it anymore.

darshana-danrox, 2019-08-25:
> Took a first look at this: it seems that while nothing on the software package side has changed too much, some of the auto-login stuff has, and e.g. the Plymouth splash screen isn't working anymore.

```
Auto log in resolved 
AUTOLOG="$(cat <<EOF
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin $SUDO_USER --noclear %I \$TERM
EOF
)"

ssh "sudo systemctl set-default multi-user.target"

ssh "sudo ln -fs /lib/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty1.service"
ssh "sudo mkdir -p /etc/systemd/system/getty@tty1.service.d && sudo touch /etc/systemd/system/getty@tty1.service.d/autologin.conf && echo '$AUTOLOG' |sudo tee /etc/systemd/system/getty@tty1.service.d/autologin.conf "  
ssh "sudo mkdir -p /etc/systemd/system/getty@tty2.service.d && sudo touch /etc/systemd/system/getty@tty2.service.d/autologin.conf && echo '$AUTOLOG' |sudo tee /etc/systemd/system/getty@tty2.service.d/autologin.conf " 
ssh "sudo mkdir -p /etc/systemd/system/getty@tty3.service.d && sudo touch /etc/systemd/system/getty@tty3.service.d/autologin.conf && echo '$AUTOLOG' |sudo tee /etc/systemd/system/getty@tty3.service.d/autologin.conf " 

# Set auto-login for TTY's 1-3
ssh "sudo ln -fs /etc/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty1.service"
ssh "sudo ln -fs /etc/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty2.service"
ssh "sudo ln -fs /etc/systemd/system/getty@.service /etc/systemd/system/getty.target.wants/getty@tty3.service"
```

jareware, 2019-08-27:
Great stuff, thanks, I'll try that as soon as I'm able!

Keep the fixes coming, if you have any. :sweat_smile:

darshana-danrox, 2019-08-28:
Thanks mate . I have generated the image too but still has some issues on my end . if you have resolve it send me config files I'll cross check and update

austinbv, 2019-08-28:
@darshana-danrox what issues are you seeing? We can create a check list here to verify 

[x] Boots directly to full-screen Chrome - with all the features of a modern browser
[x] No automatic updates - no surprises due to Chrome (or other packages) suddenly updating
[x] Automatic crash-recovery - accidental unpowering won't result in "Chrome did not shut down correctly :("
[] Custom startup graphics - displays customizable graphics instead of console messages during startup
[x] Lightweight window manager - uses Matchbox for minimal clutter and memory footprint
[x] HDMI output control - ready-made scripts for turning off the display outside of office hours, for example
[x] Cursor hiding - if you leave a mouse plugged in, the cursor is hidden after a brief period of inactivity
[x] Automatic reboots - reboots the Pi nightly, when nobody's watching, to keep it running smoothly
[x] Based on a recent Debian - if you want to add your own tweaks, all the expected packages are one apt-get away
[x] Batteries included - the most common how-to's and ProTips have been collected to the first-boot document
[] wpa_conf file works

darshana-danrox, 2019-08-30:
it was my custom testing with xsession I got it resolve mate 

Note :  on debian 10 (buster) web browser package named  

> chromium 

config file and setup script changes can be apply 

package name change chromium-browser >> chromium 
for application initiate we can apply alias  chromium-browser >> chromium  or name change directly on script  

what would be better

benderllin, 2019-09-02:
Cool! Has anyone tested already on pi4?

austinbv, 2019-09-03:
@benderllin I have it running on a pi4

Numiz, 2019-09-04:
As the entire thought behind this is to make it easy to setup and maintain, could someone actually give us an updated guide on how to configure it for RPI4 or update the .img?

jareware, 2019-09-07:
Hey folks, I appreciate the enthusiasm for this! As it stands, the Buster upgrade has broken more things than previous releases, and still takes a bit of work.

I've started the work in the [`raspbian-buster`](/futurice/chilipie-kiosk/tree/raspbian-buster) branch. Feel free to send PR's as you figure things out!

minutuslausus, 2019-09-07:
Do you have an issue list? We have a conf in 2 week where we plan on using this, so I am happy to help if you tell me what need to be solved

jareware, 2019-09-07:
I built a first alpha release off the [`raspbian-buster`](/futurice/chilipie-kiosk/tree/raspbian-buster) branch, and it's available for download here:

https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-alpha1

Please use this to do testing, and report issues here. Even better, if you can figure out fixes for them, **please include the commands you used, or which files you edited**, so they can be incorporated to the automated image build script.

@minutuslausus, see @austinbv's list above, and the things listed on the release page!

jareware, 2019-09-07:
Actually, to update @austinbv's list, I just tried putting in a `wpa_supplicant.conf` after flashing, and it worked without a hitch! Not sure what you might've done differently..?

benderllin, 2019-09-07:
2 gb model using wifi and ethernet works ok.

austinbv, 2019-09-07:
@jareware The WPA works I just hadn't confirmed it. I think with the PR everything seems to work fine.

minutuslausus, 2019-09-14:
I got the hw so I got to play with it. Almost everything worked fine, but I had to turn off the GL driver for screen rotation and overscan settings.

Other then that everything works as expected

jareware, 2019-09-16:
@minutuslausus thanks for the feedback! What do you mean specifically with "turn off the GL driver" though?

minutuslausus, 2019-09-16:
Ctrl + alt + f2 - Advanced options - A7 GL driver - use legacy driver

I believe the rasbian for rpi4 came with a new webgl driver which does not support the overcan and it's still a bit instable

For me I needed my displays rotated, and when I set it up, the pi booted in properly filling up the whole screen and rotated, but when the chrome loaded it was stuck on the upper half of the monitor

studioweb, 2019-09-18:
Hi, thanks for this update! I'm running rasberry pi 4 with 2GB memory and have trouble with my html5 video kiosk app which is looping trough vimeo mp4 files. Everything works fine on my pc / chrome but chromium throws in the 'Aw, Snap! Something went wrong while displaying this web page' message. Is there a way to search for the cause here? The web console doesn't show any error.

benderllin, 2019-09-18:
hi studioweb, marketing application?

1.Buster lite uses Adobe Flash Player 32.0.0.207 and OpenJDK 11anyway, make sure youre not running newest technology. 

2. Pi4 have enought ram for  that, but standard JavaScript and html5 usually works uni-threaded, so only one procesor will work, maybe is there a software solution, but the pi4 is not going to run all that shit easily.  

Maybe overclocking? or maybe changing how the gpu works changing the buster lite? if anyone knows...

jareware, 2019-09-19:
@benderllin is right, even though the 4 is quite powerful already, it still can't do everything a full blown PC can.

@studioweb is the page you're showing publicly available anywhere? Otherwise it'll be hard to reproduce this. :worried:

studioweb, 2019-09-19:
@benderllin This app will be used in multiple exercise room setup where a client goes trough 10 exercise rooms. In each rooms they will do a exercise based on the instruction on the monitor. Each monitor contains 2 videos and 2 webcam feeds so they can watch their own movements.

@jareware The version which goes into error after 10-15 mins is located at http://dev.studioweb.nl/webcam/test1.html (movies switch/rotate very fast on purpose) 

I'm now testing a iframe setup where the parent frame just refreshed the iframe each X minutes. It looks like this fixes the current problem I had but I'd prefer the standalone version with the JS worker since that prevent reloading the page completely. Networks checks can be done from within the worker.

IvoryDynamite, 2019-09-20:
The alpha1 release is working great for me so far, with the exception of screen resolution. I'm trying to run a 4K monitor in 1920x1080. I'm able to change the console resolution via raspi-config, but when Chromium opens we're back in 4K. Any suggestions welcome.

PS. I love this application.

kpdillon, 2019-09-20:
I echo IvoryDynamite's comments.  It's working great for me.  I installed the RPI 4 (4GB) on a new LG TV and had some resolutions issues too.  I ended up getting it right by adding these settings to /boot/config.txt 

Not sure if all these settings were needed but it is what worked for me.

disable_overscan=0
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
hdmi_group=2
hdmi_mode=4
overscan_left=24
overscan_right=24
overscan_top=24
overscan_bottom=24
config_hdmi_boost=4

jareware, 2019-09-23:
Hey, thanks a lot for these, very useful!

The core idea of the project is to _just work_ without any fiddling around in config files, or even raspi-config, so my challenge here is obviously to try and use this information to come up with a setup that works out-of-the-box on various RasPi versions AND various kinds of TV's/displays... so definitely sounds like a challenge with the 4 and/or 4K TV's. :sweat_smile:

IvoryDynamite, 2019-09-23:
Additional input if helpful:

1. I'm using a 4K monitor rather than a TV.
2. When the kiosk application starts up, it initially displays using just the upper-left quarter of the screen (so I assume 1920 x 1080 out of the 4K available). Then after a few seconds it reverts to full-screen 4K and stays there.

It probably seems silly to run in 1080p on a 4K monitor but my application uses transition effects (fades, movement, etc) and they just aren't smooth when it has to push 4K pixels around.

This application will be hugely useful to me if I can get it working so if this is the time to point me to the donation link please do so. :)

pyryk, 2019-11-21:
Just noticed that the RPI4 alpha 1 version only has a few hundred MBs of free space in `/` by default, and running `apt-get dist-upgrade` fails due to the lack of free disk space. Do you think it would make sense to grow the root partition in the image a bit to allow smoother updates?

jareware, 2019-11-28:
Hmm, I thought it had more, but you're absolutely right @pyryk, a bit more would be better!

IIRC the current size is capped so that the image will still fit onto a 4 GB SD card, but given that it's almost impossible to find even 8 GB ones anymore, optimizing for this probably doesn't make any sense!

jareware, 2019-12-04:
@darshana-danrox so you got the auto-login working again for tty's 1-3? I used the same code you did (which seems to also match [what the latest `raspi-config` does](https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L1078)), but I'm still being greeted by the login prompt on tty2 and tty3.

@IvoryDynamite I'm seeing the same behaviour, though I'm testing with a PC display; the initial page show is done in (what I assume to be 1080p) but after ~5 sec it reverts to the correct resolution for the display. I think it's some kind of delayed auto-detection.

jareware, 2019-12-04:
FYI, there's [another alpha build available now](https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-alpha2), which is based on a more recent upstream Raspbian. Still trying to find the time to fix the remaining issues. Sorry for the delay with the Pi 4; previous new models have been easier to upgrade to.