jareware/chilipie-kiosk #62: First boot works, but configuring WiFi (Ctrl+Alt+F2) "shows please" wait forever

lacksfish, 2019-07-15:
After the initial boot, chrome opens and everything looks fine.

The problem starts here: If I then press Ctrl+Alt+F2, I get stuck in an eternal "please wait" with no chance to ever set the WiFi network, etc.

Any hints/advice on this?

jareware, 2019-08-21:
This is odd. Sounds like X is also started on tty2. I wonder if it could be related to #72. :thinking:

jareware, 2020-03-13:
This issue has been had to reproduce, and we're hoping it goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware