jareware/chilipie-kiosk #63: Is there a way to enable an on-screen keyboard? (using a touchscreen)

dustinbolton, 2019-07-16:
I am using a touchscreen which works great but I'd like to display a keyboard when I tap on input fields so a physical keyboard is not required. Is there a way to do this?

jareware, 2019-07-16:
Don't recall seeing a solution off the top of my head.

Keywords you want to look for on e.g. Google would be "matchbox on screen keyboard" or some such.

If you get it working, I'd love it if you'd report back; I'll update the docs so others can benefit from your solution!

strunx, 2019-12-16:
install the Chrome Extension "Virtual Keyboard":

https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn?hl=en

and onscreen-keyboard is working....