jareware/chilipie-kiosk #64: Is there a way to disable long-press emulating a right-click?

dustinbolton, 2019-07-16:
I am using a touchscreen and when long-pressing (tap and hold down) the right-click context menu displays. Is it possible to disable this so a long-press is just seen as a normal click and hold? Thanks!

dustinbolton, 2019-07-16:
I've found some X11 config stuff about "EmulateThirdButton" but have not had luck in figuring out how to modify this setting.

jareware, 2019-07-16:
Similar to #63, sadly I don't recall seeing such a solution yet, but please report back if you find one!

allcoolusernamesaregone, 2019-12-20:
Maybe try the chrome web store addon "Context Menu Blocker". This works fine for disabling mouse right clicks, and I think it should help with a touch screen also.

allcoolusernamesaregone, 2020-02-21:
> Maybe try the chrome web store addon "Context Menu Blocker". This works fine for disabling mouse right clicks, and I think it should help with a touch screen also.

As tested so far: Context Menu Blocker works with a touch screen. the kiosk startup flag for chrome is recommended also

jareware, 2020-02-24:
I would assume the interpretation of a long-press as right-click is done by the drivers of whatever touch screen you're using.

Which one are you using, btw?

allcoolusernamesaregone, 2020-02-24:
i'm using the Iiyama T2235MSC-B1 Touchscreen-Monitor. working great, somehow TOO great - raspberry 3 btw.
top great means: Multitouch is working. so i needed some effort and disabling/preventing Scripts in the displayed Website, as no Zoom in/Out or other stuff shall Happen. if interested i can Share here.

allcoolusernamesaregone, 2020-02-25:
Hi,

here for whom it may interest my modifications of the kiosk system.
Usage is a kiosk with printer, the displayed website is always the same (with some time etc dependent content modification, but mainly the same site) with some buttons on the touch screen which cause the printer to print a ticket (queuing system).

As I use a touch screen I do not want the user to have the possibility to have multitouch or touch navigation or select/highlight text etc.
And I need the possibility to control the displayed site (for my usage: change query parameters in the url) from the server.

So using the plugin
https://chrome.google.com/webstore/detail/chromix-too/ppapdfccnamacakfkpfmpfnefpeajboj?hl=en-GB
for remote control and the plugin
https://chrome.google.com/webstore/detail/context-menu-blocker/hibfjefoddeenakbenbegknbhfhndjbj
to disable context menu.

in the displayed website:
/* Javascript in shown Kiosk website to prevent multitouch / zoom gestures */
//disable select/mark text
document.onmousedown=function() {  return false; }
document.onselectstart = function() { return false; }
//disable right click
window.addEventListener("contextmenu", function(e) { e.preventDefault(); })
//disable multi touch touch stuff
window.addEventListener("touchstart", touchHandler,  { passive: false, capture: false, once: false });
function touchHandler(event) {
  if (event.touches.length > 1) {
    //the event is multi-touch
    event.preventDefault();
    event.stopImmediatePropagation();
    return false;
  }
}

Modification of chrome flags to work with touch screen and prevent unwanted user actions (back, forth navigation, exit fullscreen, and enable auto audio play without previous user action):

-	Overscroll history navigation = disabled #disable back & forth navigation
-	Experimental fullscreen exit UI = disabled #disable pull top to buttom gesture to show “an Exit X Bubble” to touch
-	Autoplay policy = no user gesture is required

one time Modification of /etc/fstab: set various directories to a ram disk: do not write to sd card – extend lifetime
#tmp and working dir for the background java server
tmpfs /mnt/RAMDISK tmpfs nodev,nosuid,size=32M 0 0  
#std directories to tmp
tmpfs /tmp tmpfs defaults,noatime,size=4M 0 0
tmpfs /var/tmp tmpfs defaults,noatime,size=4M 0 0
tmpfs /var/log tmpfs defaults,noatime,size=32M 0 0

one time modification in a shell: disable swap. goal: do not write to sd card – extend lifetime
sudo dphys-swapfile swapoff
sudo dphys-swapfile uninstall
sudo update-rc.d dphys-swapfile remove

Modification of .xsession for modification per startup:

rm /home/pi/.xsession-errors   #clean the file for a new session - I start e.g. the server application in via the .xsession and the server logs (if not using >/dev/null...) to the .xsession-.errors file which grows limitless..

#a previously clean copy of the profile directory backuped in CHROMIUMCONFIG
#cp this "config template" to the ramdisk and use it for chromium: goal: chromium does not write to the sd card!
cp -r /home/pi/aufruf/CHROMIUMCONFIG /mnt/RAMDISK/.config  
#enable kiosk mode 
chromium-browser --user-data-dir=/mnt/RAMDISK/.config/chromium --kiosk --start-fullscreen --window-size=1920,1080 --disable-infobars

May it help :)

jareware, 2020-02-29:
Thanks a lot for the thorough explanation!

I think we would do well to incorporate this to some docs at least.

mbretter, 2020-03-18:
The longpress as right mouse click "feature" is a chrome issue, it blocks the touch longpress event on my application, I switched over to firefox, which works well in this case.

blecx, 2020-06-13:
Well, long tab is emulating the right mouse click, that triggers the contextmenu. The contextmenu itself is an event and you can bind an event handler to it

Like from@allcoolusernamesaregone 
//disable right click
window.addEventListener("contextmenu", function(e) { e.preventDefault(); })

or in reactjs
 onContextMenu={(e) => {
        //this prevents righ-click contentmenue event on long tab to pop up the menu
        e.preventDefault();
      }}
This work around was tested by me. If you try a e.preventDefault() on touch start or mouse dow/mouse click event. It will not stop propagating the event. So this will not work. A also tested this.

Hope this.

dave-watts, 2023-10-12:
This work around doesn't work because of a 

"Unable to preventDefault inside passive event listener due to target being treated as passive"

error