jareware/chilipie-kiosk #71: How to change start URL via SSH?

OZ1SEJ, 2019-08-16:
Via keyboard, I've pointed chromium to my desired web page, and now it's running just fine. But I know that at some point, I will need to point chromium to a different address. Can't I do this via SSH, or will I need to plug in a keyboard to do that?

jareware, 2019-08-16:
Please check out #38, and chime in to the discussion if you have additional ideas!

closed 2019-08-16 by jareware

OZ1SEJ, 2019-08-16:
I gather that this is not possible with chromium - I've installed VNC instead, so I can do it remotely. Thank you!

matristain, 2019-08-16:
I´ve used something like this:

bash -c "export DISPLAY=:0; xdotool key F11 sleep 1 key ctrl+l sleep 1 type 'https://google.com'; xdotool sleep 1 key KP_Enter; xdotool key F11;"

Not very pro, bud useful.

jareware, 2019-08-17:
@matristain that's exactly the kind of solution that many people are looking for!

I'll add it to the README if you don't mind. :smile: