jareware/chilipie-kiosk #72: Starts in raspi-config after nightly reboot

OZ1SEJ, 2019-08-21:
I've tested my chilipie-kiosk thoroughly, and after running it on my desk for a few days without problems, I put it on our infoscreens. Now, after the first nightly reboot, it apparently booted directly into raspi-config:

[image]

How would I go about trying to figure out what went wrong?

jareware, 2019-08-21:
Hmm, interesting! For better or worse, you're not the only one to report this (see issue ref above), though I have no idea how common this is.

Have you seen this happen consistently? Did the kiosk always boot into tty1 on the old display, and always to tty2 on the infoscreen display? That might give us hints as to debugging this. :thinking:

OZ1SEJ, 2019-08-21:
No, it's never done this on the other display. Both displays are VGA, so I have an HDMI-VGA adapter between the pi and the screen, and those are two different adapters. Do you know if there is a relevant log file I could look at?

OZ1SEJ, 2019-08-22:
Ok, now I've been able to play with it a bit:

1. When ./display-off.sh is run, the display turns off immediately.
2. When ./display-on.sh is rum, the display turns on immediately, showing raspi-config.

My solution is to just reboot in the morning instead of running display-on.sh. Boots up fine.

jareware, 2019-08-22:
@OZ1SEJ thanks a lot for getting to the bottom of this, super useful info! :tada: 

We'll make sure we do more testing around this for the next release.

jareware, 2020-03-13:
This issue has been had to reproduce, and we're hoping it goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware