jareware/chilipie-kiosk #73: Chromium process stop after some hours.

matristain, 2019-08-22:
I have this issue just in one device I so fat the info I have is:
This device does not pass the 65-67 top temperatures (normally it's about 60)
It happens after about 4-5 hour of playing some images end video.
First I used to reboot the device to get chromium working again
Now i run the .xsession file every time I realize chromium stop working.

PS:
I also try to overtake this issue with something like this on the crontab:

* * * * * export DISPLAY=:0 && pgrep chromium-browse > /dev/null || ./.xsession

*This does not solve the issue but at least the device keeps doing what it is meant to do.

kpdillon, 2019-11-04:
I think I might be hitting this bug.  Was a new version of chromium applied to the beta for RPI 4?

BasilHall, 2019-11-07:
We were still experiencing issues with both YouTube and local videos on Chromium for the first release or two of Raspbian Buster. I think there were some issues with the custom patching that Raspbian does to Chromium. 

We are running the latest Raspbian release now (fully patched) and it seems to be better, but we still saw a glitch the other day that I am working to confirm was related to playing local HTML5 videos specifically. We do not appear to be having any more issues with embedded YouTube videos. 

The beta release of Chilipie-kiosk is still built on one of the older Raspbian Buster releases, and it wouldn't surprise me if you are running into some of those earlier issues. A lot of work still seems to be going on in the mainline Rasbian distro for both video drivers and Chromium optimization, so I expect to see this continue to evolve for a bit.

kpdillon, 2019-11-07:
Good info!  Any chance of getting a new alpha build soon?

BasilHall, 2019-11-07:
I should clarify that I am not one of the Chilipie-Kiosk devs, but within my company we have been using it on RPI 3's for a while now internally. 

We are in the process of trying to move to using RPI 4's in all of our new instances, so we are wading through the hurdles as we come across them and I am trying to share what we are finding as we go.

studioweb, 2019-11-24:
@BasilHall We're also experiencing similiar issues where 3 out of 8 raspberries (4s withb4gb) would get stuck (still image, keyboard would mot work anymore) while playing HTML5 video's from Vimeo. Planner reboots would mot solve this unfortunalety. Because of a deadline we're forced to choose for Intel HDMI stick with Windows 10. Chrome and Edge work very Stable without these kind of issues.Hopefully these issues will be solved.

See also this topic: https://github.com/futurice/chilipie-kiosk/issues/61#issuecomment-533008398

jareware, 2019-11-28:
> Good info!  Any chance of getting a new alpha build soon?

Yes!

jareware, 2019-12-04:
FYI, there's [another alpha build available now](https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-alpha2), which is based on a more recent upstream Raspbian. You may want to see if it has fixed this issue. Sorry for the delay with the Pi 4; previous new models have been easier to upgrade to.

kadusalles, 2020-01-27:
Hi guys, maybe i was struggling with the same problem. My chromium screen goes with after some hours of video, like if there is nothing on the page.

jareware, 2020-02-04:
All this sounds like there might still be some issues with some graphics drivers that ship for the 4. :neutral_face:

jareware, 2020-03-10:
I'm closing this issue for now, since it seems to have to do with upstream Raspbian we have little control over. Also hoping it's (already?) fixed by newer Raspbian releases.

There's a new chilipie-kiosk release coming soon, where this can be tested. Please reopen if you keep seeing this.

Thanks for all the feedback!

closed 2020-03-10 by jareware