jareware/chilipie-kiosk #78: Controlling dual monitors on RPI4

Numiz, 2019-09-10:
It would be fantastic if there's a way to control two independent screens. 
Scenario: We've got two 4k monitors hooked up side by side to use for our operations department at work. The RPI4 is able to use 2x4K monitors at 30hz which is more than enough to handle semi-static pages such as Jira Boards. 

Is there a way this is doable?

Numiz, 2019-09-13:
Hi @jareware, I'm currently running the Alpha on our RPI4 4GB, and it's working wonders! However when plugging in both monitors it sees both monitors as 1, so the splashscreen & all webpages are stuck in the middle of them :D

jareware, 2019-09-13:
Oh, interesting. :)

You'll probably want to google around for "xorg dual screen" something.

Let us know if you come up with a working config!

jareware, 2019-12-04:
FYI, there's [another alpha build available now](https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-alpha2), which is based on a more recent upstream Raspbian. You may want to see if it has fixed this issue. Sorry for the delay with the Pi 4; previous new models have been easier to upgrade to.

Numiz, 2019-12-04:
Interesting! Will take this for a test tomorrow and report back to you!

Numiz, 2019-12-11:
Hi @jareware, I've checked now with V3.0.0-alpha2 but the problem still persists. Splash screen is between the monitors and everything is seen as one entire monitor. 

Raspbian 4.19 does not have these issues. 
Are you able to provide me with any ideas?

jareware, 2019-12-12:
OK, I think that has more to do how X.org treats multiple monitors - effectively as one big monitor, which is just oddly shaped.

Don't have a solution off the top of my head, sorry!

jareware, 2020-03-13:
We're hoping this issue goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware