jareware/chilipie-kiosk #79: Session cleanup issue

crisp2u, 2019-09-17:
Hello,

After playing a bit with this wonderful tool I noticed that for me the previous session cleanup that is done here

https://github.com/futurice/chilipie-kiosk/blob/818a81335f8ec75e367b60339d1a481613e5db63/home/.xsession#L29

is not working. `rm` is passed a `Last Session` file that contains a space and fails.

I tried manually running this with `find .config/chromium/ -name "Last *" -exec rm {} +` and is looking good.

Anyone else experiencing this ? 

I'm running the v2.1.0

jareware, 2019-09-18:
Thanks for reporting this! It indeed looks like it might not work as intended.

Possible reason for why no-one's noticed is that it's actually not needed all that often, Chromium recovers fine on its own most of the time.

So this manifested as the "Chrome didn't shut down correctly :(" page after a reboot?

crisp2u, 2019-09-18:
No, in my case It was a lot of tabs opened after restarts. I actually realized this when testing the tab cycle feature between two tabs opened from the command line

matristain, 2019-09-18:
I added this to my .xsession before  chromium-browser --start-fullscreen...

`sudo rm -r ~/.cache/chromium`
`sudo rm -r ~/.config/chromium/Default/`

Refer to issue [47](https://github.com/futurice/chilipie-kiosk/issues/47) to see why I used this for.

jareware, 2019-09-19:
@crisp2u hmm, Chromium IS configured to reopen any tabs that were open when it shut down (or crashed)... do you mean there were more than that when it reopened?

crisp2u, 2019-09-22:
@jareware Yes, I had quite a few tabs opened. I open 3 tabs by passing 3 urls as arguments to chromium-browse in ~/.xsession. I believe that was the intention of the above line in ~/.xsession, to delete the chromium Last Session but that fails, like I explained above

crisp2u, 2019-09-22:
added a PR #80

jareware, 2019-09-23:
OK, thanks, I'll check that out ASAP!