jareware/chilipie-kiosk #82: Translate dialog visible after reboot

OZ1SEJ, 2019-09-28:
Yesterday, after a reboot, I suddenly had a "Translate this page?" dialog open. I closed it over VNC, but I was somewhat surprised to see it, as I haven't seen it before with chilipie-kiosk. I haven't tried to recreate it yet, but I'll know on monday.
[image]

jareware, 2019-09-30:
OK, thanks for reporting, this definitely shouldn't pop up.

Can you tell me what your chilipie-kiosk version is (if in doubt, it's in a file called `.chilipie-kiosk-version` on the pi home directory) and what your Chromium version is (go to `about:version` on the pi). This might help us nail down the issue.

OZ1SEJ, 2019-09-30:
Chilipie-kiosk version: 2.1.0
Chromium version: 
```
Chromium	72.0.3626.121 (Official Build) Built on Raspbian , running on Raspbian 9.4 (32-bit)
Revision	da3787ba355f18db7db52abf75c42afb408d656f-refs/branch-heads/3626@{#883}
OS	        Linux
JavaScript	V8 7.2.502.28
Flash	        (Disabled)
User Agent	Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Raspbian Chromium/72.0.3626.121 Chrome/72.0.3626.121 Safari/537.36
Command Line	/usr/lib/chromium-browser/chromium-browser-v7 --enable-pinch --start-fullscreen --window-size=1920,1080 --disable-infobars --autoplay-policy=no-user-gesture-required --flag-switches-begin --flag-switches-end
Executable Path	/usr/lib/chromium-browser/chromium-browser-v7
Profile Path	/home/pi/.config/chromium/Default
```
I'm a little surprised by the --window-size setting, since my screen is 1360x768, and that is specified in /boot/config.txt with HDMIGROUP and HDMIMODE, and it works. So it seems that that settings is ignored - lucky in this case :-)

jareware, 2020-03-10:
I'm building the next release of the kiosk, and I tested for this specifically by navigating to www.baidu.com, reloading, and even rebooting. I couldn't conjure up the "Translate this page" dialog.

Clearly you did, but I'm willing to settle this as that just being an older version.

Sit tight, new release coming soon!

closed 2020-03-10 by jareware