jareware/chilipie-kiosk #85: Pi Zero won't boot without hdmi signal.

pointtom, 2019-10-17:
Love the project, works well on my monitor, boots no worries. into 1080P
However connected to a Television via hdmi it gets as far as your logo then hangs forever.
I then tried having the TV turned off, same result.
Then tested the pi without a hdmi signal at all.
Hangs as well.
Here is my config settings..
[config.txt](https://github.com/futurice/chilipie-kiosk/files/3737133/config.txt)

Tom Duncan

jareware, 2019-10-17:
Hmm, so do I understand correctly that your Pi Zero boots up when connected to a PC monitor, but hangs when connected to a TV?

pointtom, 2019-10-17:
Yes, correct. Also if you don't have an HDMI connection at all.

jareware, 2019-12-04:
FYI, there's [another alpha build available now](https://github.com/futurice/chilipie-kiosk/releases/tag/v3.0.0-alpha2), which is based on a more recent upstream Raspbian. You may want to see if it has fixed this issue. Sorry for the delay with the Pi 4; previous new models have been easier to upgrade to.

rvdleun, 2019-12-30:
I had a similar issue as above(connected to a television, remained stuck on the 'Please Wait' page). After installing the alpha build, it's working now.

jareware, 2020-03-13:
This issue has been had to reproduce, and we're hoping it goes away with the upcoming 3.0 release, as it's based on a much newer upstream Raspbian. Release Candidate builds of 3.0 are already available for download in [releases](https://github.com/futurice/chilipie-kiosk/releases).

Thus, I'll be closing this issue for now. Feel free to reopen if it seems like this issue persists.

closed 2020-03-13 by jareware