jareware/chilipie-kiosk #89: Etcher Windows Install

alexcclarke, 2019-11-05:
I download the image .img.tz and unzip with 7zip as a .img when i load the image into etcher it give me a warning 'It looks like this is not a bootable image.  This image does not appear to contain a partition table, and might not be recognized or bootable by your device.'

When i click continue the flash is successful but the pi will not boot.  Any ideas?

jareware, 2019-11-28:
Interesting.

I don't have easy access to a Windows box on which I could try flashing it, but I was under the impression that Etcher would work the same there too.

Just to make sure, by `.tz` you mean `.tar.gz` from the [Releases page](https://github.com/futurice/chilipie-kiosk/releases)? Have you made sure that the SHA1 matches?

Filmore85, 2020-02-26:
I am having the same issue. I install the img.tar.gz and Etcher says it is not a supported image type

alexcclarke, 2020-02-27:
BTW I never got this sorted so I opted for another solution.

allcoolusernamesaregone, 2020-02-27:
I'm using win32 disk imager (https://sourceforge.net/projects/win32diskimager/), never had issues here

wujood, 2020-07-24:
For me it got solved when i unzipped the tar.gz and used the plain .img file. I used WinRar to unpack it.

jareware, 2020-10-08:
I'm closing this now due to lack of activity, and people also reporting being successful.

Feel free to reopen if this issue persist.

birdybro, 2021-05-11:
This wasn't closed by the way, and the issue persists. You have to unzip the tar.gz to resolve it. The image is not setup correctly for use with Etcher, when flashed according to the instructions, it produces no boot partition, and you can't edit anything, just sees it as an unformatted partition for the whole SD Card. This was done in Windows 10. Both Raspberry Pi Imager and balenaEtcher produced the same result.

then I unzip img.tar from the img.tar.gz, then I have to unzip img.tar as well (used 7zip in windows for all of these steps), so I can get chilipie-kiosk-2.1.0.img.

After that I finally can flash the image to the microsd and the result is....

success!

So unless the user fully unzips the .img file nested inside the gzip nested inside the tar archive... windows users will not be able to flash this successfully. Also gets a black screen on the pi and doesn't work unless you go through all this.

Something is incorrect about the way the .img file is being archived initially. I know that on a Mac I think it was either tar or gzip that was under the BSD license whereas on linux it's under GPL, and they have some differences from each other. It was one or the other, can't remember which, that might potentially be having an impact.

jareware, 2021-06-16:
Thanks a lot for figuring this out!

Need to do something about the packaging, but in the meanwhile, better instructions much appreciated!

closed 2021-06-16 by jareware