jareware/chilipie-kiosk #90: Clean install, no space for upgrades

trecords, 2019-11-09:
Hi, I have 16GB memory card on raspberry PI 3B also 7 inch screen attached to RPi. So first I burned chilipie-kiosk then installed openhab seemed something wrong and couldnt make HA to start. then burned chilipie-kiosk again and just run apt-get upgrade, so I noticed there are no enough space left.

df- H shows 2.44GB used by chilipie-kiosk and 12GB is unallocated. Do you have any solution for this ?

Regards,
Turkel.

matristain, 2019-11-09:
Hi trecords
Just need to Expand filesystem :
$ sudo raspi-config --expand-rootfs

The image is just the right size to burn chilipie-kiosk , so after flashing the first ting to do is expand the file system to get microSD full size available.

trecords, 2019-11-09:
into which directories should I allocated this unallocated space? This is how it looks like in my standard rasbian

`/dev/root        16G  2.3G   13G  16% /
devtmpfs        481M     0  481M   0% /dev
tmpfs           486M  2.1M  484M   1% /dev/shm
tmpfs           486M  6.5M  480M   2% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs           486M     0  486M   0% /sys/fs/cgroup
/dev/mmcblk0p1  265M   55M  210M  21% /boot
tmpfs            98M     0   98M   0% /run/user/1000
`

jareware, 2019-11-28:
@matristain is exactly right!

Though maybe the instructions for doing that should be made more prominent in the docs...