jareware/chilipie-kiosk #92: How to rotate screen with Pi4 buster ?

yanisla35, 2019-11-16:
HI,

I'm looking for a way to rotate screen with the PI4, but there is no way to use 'display_rotate' because of "vc4-fkms-v3d'. But I really need this driver to run... 

Then, what is the solution to rotate the screen without deactivating "vc4-fkms-v3d" ?

Thank you in advance.

jareware, 2019-11-28:
I wasn't aware of the rotation not working on Pi 4.

I'd be happy to hear of any solution you might come up with!

jareware, 2019-12-04:
https://github.com/futurice/chilipie-kiosk/issues/61#issuecomment-531467096 seems related.

jareware, 2020-03-10:
Can confirm that you need to disable the line

```
dtoverlay=vc4-fkms-v3d
```

in `/boot/config.txt` if you want this to work on Rpi4.

jareware, 2020-03-10:
I'll update the relevant part of the first-boot doc at least. Better than nothing.

wujood, 2020-08-01:
I just found out that you do not need to disable dtoverlay by inserting a line into the `.xsession` file
```
xrandr --output HDMI-1 --rotate left
```
I tried a lot of things, but this seems to work. Tried it on a fresh install without changing anything else. Here is how the beginning of the file looks:

```
#!/bin/bash

export DISPLAY=:0.0

# Start cursor at the top-left corner, as opposed to the default of dead-center
# (so it doesn't accidentally trigger hover styles on elements on the page)
xdotool mousemove 0 0

# Set some useful X preferences
xset s off # don't activate screensaver
xset -dpms # disable DPMS (Energy Star) features.
xset s noblank # don't blank the video device

# Rotate screen 90 degrees
xrandr --output HDMI-1 --rotate left

# Set X screen background
sudo nitrogen --set-centered background.png

# Hide cursor afer 5 seconds of inactivity
unclutter -idle 5 -root &
[...]
```

@jareware If this could make it into the docs, that would be awesome, because i spent days until i figured this out.

jareware, 2020-10-08:
Definitely! And feel free to send a PR for the docs if you have time!

marutimuthu, 2020-12-03:
> # Rotate screen 90 degrees
> xrandr --output HDMI-1 --rotate left

Tried adding ^ to .xsession file
It did not work. Is there any other configuration required ?

wujood, 2020-12-03:
Could you provide more context? Where did you add the line? Did you change anything else?

marutimuthu, 2020-12-04:
I added the line in the .xsession file. No other changes. 
This worked for me:
xrandr --rotate left

wujood, 2020-12-04:
Does it also work if you use HDMI-2 as output? There are two outputs on the raspberry so you would need to change that accordingly

marutimuthu, 2020-12-05:
I guess for that you will have to `export DISPLAY=:0.1` corresponding to HDMI-2