jareware/chilipie-kiosk #94: Unhandeld prefetch abort makes chromium show white page

DerMolly, 2019-12-02:
I'm using chiliepie-kiosk for about 3 weeks now. Unfourtnatly chromium shows an white page at unregular intervals. I could also extract this error message from `journalctl`.

`kernel: Unhandled prefetch abort: breakpoint debug exception (0x002) at 0x025029d0`

The kernel message seems to indicated some problem with a 32-bit on a 64-bit https://bugzilla.redhat.com/show_bug.cgi?id=1526618

Any idea what I can do to fix this problem?

jareware, 2019-12-03:
That's... interesting.

Which version of the image are you using on which hardware?

DerMolly, 2019-12-03:
Sorry I forget to mention that:

Image Version:  tag v3.0.0-alpha1
```
Linux $name 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```
Hardware: RPi 3B+
```
$grep Revision /proc/cpuinfo
Revision        : a020d3
```

jareware, 2019-12-04:
The Chromium that's used is the one officially supported by Raspbian, so it _should_ be suitable...

What kind of a page are you showing?

jareware, 2019-12-04:
(Accidentally the wrong button)

DerMolly, 2019-12-04:
infoscreen.oh14.de
It‘s a small dashboard for your student room with a few infos. Until a few weeks it worked quite well on an old EEEBox, so I don‘t think the dashboard is the culprit.

jareware, 2019-12-05:
Definitely not "the culprit" as in it's doing something wrong, but just wondering what might be different about it that's causing Chrome to throw in the towel. EeeBoxes are x86 and the Pi is ARM, so they're not necessarily comparable.

I'd probably either:

* wait for Raspbian to update their Chromium in their repos
* manually build a newer one
* check if a simpler page behaves the same, and if not, try changing the dashboard somehow

I realize none of these options are very appealing, but it's all I can come up with right now.

kpdillon, 2020-03-10:
When I get a white screen (full or partial) this is how I have worked around it, by running this script in cron every minute.

#!/bin/bash

thecpu=`top -b -n1 | grep "Cpu(s)" | awk '{print $2 + $4}'`
#echo $thecpu

if [ ${thecpu%%.*} -le 3 ]; then 
	export XAUTHORITY=/home/pi/.Xauthority
	export DISPLAY=:0.0 && xdotool key CTRL+R
	echo `date` "The CPU is $thecpu which is less than 3% - running refresh" >> /home/pi/fix.log
else
	echo "The CPU is $thecpu."
fi

jareware, 2020-03-10:
Closing this as there's not a lot we can do.

Please reopen if you still hit this issue. Hoping it'll be gone in the next release.

closed 2020-03-10 by jareware