jareware/chilipie-kiosk #97: Feature request

Madseason41, 2020-01-16:
Is there any chance you could build in the following feature?

https://github.com/RebelIT/rpIoT

Also the ability to configure chrome (or at least the load url) via a web server would be a fantastic.

Thanks and great work!

allcoolusernamesaregone, 2020-02-21:
Have a look at chromix-too: https://github.com/smblott-github/chromix-too/
Working great for me. 

needs an addon in chrome: https://chrome.google.com/webstore/detail/chromix-too/ppapdfccnamacakfkpfmpfnefpeajboj?hl=en-GB

and a little server application running (chromix-too-server)

plugin connects to the server by websocket, and the chromix-.too client tool connects to the server also and communicate over the server to the plugin and chrome itself.

e.g. do a click on an element as an example from my app 
(from a bash script, $1 = element id)
chromix-too raw chrome.tabs.executeScript '{"code":"document.getElementById(\"'$1'\").click();"}'

with "chromix-too raw" you can do anything as a chrome addon itself could do.

easier example: get the active url (url shown in the current tab/Window)
chromix-too raw chrome.tabs.query '{"active": true}' | jq '.[].url'

And your wish: load url
chromix-too raw chrome.tabs.update '{"url":"http://....'"}'

jareware, 2020-02-24:
Closing as dupe of #38.

Thanks for the detailed info on your `chromix-too` setup @allcoolusernamesaregone, we'll be sure to consider that as the practical implementation for #38 when the time comes!

closed 2020-02-24 by jareware