jareware/chilipie-kiosk #99: Chromium shows a popup "Can't update Chromium..."

jmetelski, 2020-02-12:
Hello!

I have chilipie version 3.0.0, updated with apt to the current date (feb 12 2020). I'm testing the OS before deploying it in production.

Chromium displays a popup saying that it can't be updated (sorry, I don't have the literal words in english of the message, because I have the system in spanish language) and a button labeled "Reinstall Chromium" and of course an "X" to close the popup.

I closed the popup a day ago, but in the night it showed up again. 

Do you know any way to disable this message?

pi@chilipie-kiosk:~$ chromium-browser --version
Chromium 78.0.3904.108 Built on Raspbian , running on Raspbian 10

jareware, 2020-02-13:
Hey, that's definitely new.

Can you tell me which exact `apt` command you used?

jmetelski, 2020-02-13:
Hi, thanks for your reply!

Yes, I used
`apt update`
and
`apt dist-upgrade`

I googled and I found people having this issue on other systems, for example:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=943668

but found no useful workaround or solution.

Thanks!

jareware, 2020-02-18:
The thing is, chilipie-kiosk is attempting to work more like an appliance - in that if you want to upgrade, you should just flash a newer image.

See #11.

Hope this helps!

jmetelski, 2020-02-18:
OK, I'm really sorry to have bothered you, I didn't know. I'll just see if I have the latest version, start fresh with that one and avoid any upgrades. Thanks!

jareware, 2020-02-24:
No worries! Part of the issue I linked is indeed coming up with a better way to communicate this philosophy to users. :wink:

jareware, 2020-03-10:
Working on the 3.x builds again, and I'm seeing this too.

Reopening.

jareware, 2020-03-10:
Seems this is a fairly common issue: https://www.raspberrypi.org/forums/viewtopic.php?t=264399, https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=943668, https://askubuntu.com/questions/1196771/cant-update-chromium-issue, etc...

blueicehaller, 2020-05-22:
Hello,
I'm running [Raspberry Pi 4 support (RC2)](https://github.com/futurice/chilipie-kiosk/releases) on an Raspberry Pi 4.

When I exit the Full Screen (F11) it shows that there is an Update for Chromium and the Option to Restart Chromium. But this doesn't work.

Even if the Update for Chromium got updated via apt-get it seem that I would need to run Chromium with root rights to Update it.

How should I proceed to get the automatic the latest stable release of Chromium?

Thank you in advance.

codeclinic, 2020-06-02:
> The thing is, chilipie-kiosk is attempting to work more like an appliance - in that if you want to upgrade, you should just flash a newer image.

The problem with this is that you can still see available upgrades and upgrade the system normally via apt. Also, I'm sure there are some others like myself that have other scripts / services running on the Chillipie system. In this case it'd be so useful to have an upgrade script that could upgrade the chilipie software so as to not have to recreate all those scripts & services.

jareware, 2020-08-24:
@codeclinic yeah that's indeed not ideal, having to re-flash if you have other stuff running on the same Pi.

The thing is though, since folks can do pretty much anything to customize the OS after installation, being able to reliably upgrade it would be really challenging..!