jareware/chilipie-kiosk #100: Same ssh host key on all images?

sualko, 2020-02-25:
Hi,

I was looking at the image setup script and it looks like the ssh host key is not deleted, this means someone who has enabled ssh can not be sure that he is always connected to the same machine. The risk is medium, because this kiosk system is most of the time probably not reachable via the internet, but a notice in the docs would be great.

An instruction on how to regenerate host keys can be found on https://www.cyberciti.biz/faq/howto-regenerate-openssh-host-keys/.

Cheers

jareware, 2020-02-25:
Hey,

Thanks for paying attention to security, these tips are always welcome!

In addition to the systems usually not being reachable over the internet, the temporary SSH key is in fact removed by the script:

https://github.com/futurice/chilipie-kiosk/blob/818a81335f8ec75e367b60339d1a481613e5db63/docs/image-setup.sh#L178

Similarly, SSH is actually disabled entirely, until the user specifically opts into re-enabling it.

sualko, 2020-02-25:
I'm talking about the host key which is used to identify the server. Not the client key. If someone knows the host key, he could fake the server, aka man in the middle. I think the risk that someone is interfering the connection is low, but the fix should be easy, so maybe I create a pr later.

jareware, 2020-02-29:
Ah, sorry, read your original comment in a hurry! That's indeed the case.

You're right, regenerating the key on first boot might be a good idea!

sualko, 2020-02-29:
I need to verify this, but it should be enough to enable the `regenerate_ssh_host_keys` service. 

```
/bin/systemctl enable regenerate_ssh_host_keys
```

I will create a pr after I gave it a try.

jareware, 2020-03-06:
Merged, thanks a lot! :bowing_man:

PR #105 (2020-03-06): remove ssh host keys and enable regeneration
fix #100
commit c0bc6003 (2020-03-06): remove ssh host keys and enable regeneration

fix #100
closed 2020-03-06 by jareware