jareware/chilipie-kiosk #104: Include ntp in base image?

SnorreSelmer, 2020-02-28:
The rPi doesn't have a RTC backup battery, and the clock slips a little bit every day. HTTPS has a security feature that if the requester's system-time is off by too much, HTTPS websites won't load.
With ntp pre-installed, system-time would always be very accurate.

jareware, 2020-02-29:
Honestly I was under the impression Raspbian already shipped with one... :thinking: but I'll take your word for it (can't check right now).

PR would be welcome!

SnorreSelmer, 2020-03-01:
sudo apt installerer ntp installs two packages, so apparently not. :-)

allcoolusernamesaregone, 2020-03-01:
the ntp client was replaced with systemd-timesyncd in raapbian strech i think.
there should be no need for ntp If one just need the correct time after a reboot...?

jareware, 2020-03-03:
I can't say I've noticed the clock slipping on any of mine, so I'd be inclined to assume what @allcoolusernamesaregone said.

SnorreSelmer, 2020-03-03:
Ah ok. Must be some networking issue regarding the NTP protocol on our end
then. *sigh* _Another_ thing I need to bug the networking/firewall people
about. Thanks for the help though!

jareware, 2020-03-06:
np!

closed 2020-03-06 by jareware