jareware/chilipie-kiosk #106: Wifi Issues

Filmore85, 2020-03-04:
Hello, I am having this issue on both 3.0 and 2.1

After using etcher to flash the image file to the SD card, I remount the SD and there is no boot folder, the SD card is renamed to boot, it is full of files, has an overlays folder and a hidden .fseventsd folder. I copy the WPA_Supplicant file over to the SD, not into any folder but with the other files. after booting it doesnt connect to wifi and when i remount the SD card, the WPA_Supplicant file is gone. using ctrl+alt+f2 and adding in the wifi that way also does not work.

Filmore85, 2020-03-04:
Using win32 disk imager instead of etcher resolved this issue

Filmore85, 2020-03-05:
I used win 32 disk imager and encountered the same issue

jareware, 2020-03-10:
>  using ctrl+alt+f2 and adding in the wifi that way also does not work

I think that's the big clue: I would say there's something wrong in the WiFi config itself.

I'm closing this as I don't think there's anything we can do.

closed 2020-03-10 by jareware

Filmore85, 2020-03-23:
The issue was that I was trying to connect to a Hidden Wifi network, just needed to add  scan_ssid=1
to the wpa_supplicant within the brackets