jareware/chilipie-kiosk #112: How to enable Onscreen Keyboard for Touchscreen (HDMI, USB)

blueicehaller, 2020-04-27:
Hi,
on an Raspberry Pi 4 I connected the follwoing Display with Touchscreen:
Iiyama ProLite T2735MSC

Is it possible to enable Onscreen Keyboard?

Thank you in advance.

jareware, 2020-08-24:
Sorry for taking a long time to get back to this.

There's no built-in support for showing 2 sites at the same time, unfortunately.

As to white-listing, you can definitely do something on the system level (or maybe using some Chrome extension?). That said, the kiosk isn't really hardened for adversarial users - if someone has physical access, they can probably break out of the browser.

closed 2020-08-24 by jareware

blueicehaller, 2020-08-24:
That's OK. I updated my requirements with my current situation.

jareware, 2020-08-25:
Thanks for detailing your setup, might be useful for someone looking to do the same!