jareware/chilipie-kiosk #113: How can i hide the Mouse-Arrow on a Touchscreen

nighteagle1974, 2020-05-02:
Hello,

so the .xsession is in the image not under /home/ - it is under home/pi/
What is wrong -- the documentation?

So next i want to have unclutter -idle 0 but the Mouser-arrow is still one second if i touch on the touchscreen. Is there a way to hide the mouse-arrow on all conditions? Or have i install a cursor-theme with an minimal point-mouse cursor or a transparent-mouse-cursor.
I have a touchscreen i don't want see the mouse-arrow. I have Buttons on the Touchscreen...

Some ideas??

gitroli, 2020-05-03:
for me its the same

allcoolusernamesaregone, 2020-05-03:
do you have control over the shown Content? If, maybe just add

    <style>
      html {
        cursor: none; 
      }
    ....

to your Website?

nighteagle1974, 2020-05-04:
I use ioBroker and VIS so the visulation ist little bit confuse... but in other applications oftern you dont have access to the content - only a web-address - so in this case is there a way to deativate
Because on an old raspian install with chromium i have used unclutter and some other things to deactivate it and it works. But here is an X11 server not the other modern ones?

nighteagle1974, 2020-05-05:
Looks like it is not a good solution for kiosk - there is no complete documentation and noboy can help with this chilipie.

Solution:
I have install fresh actually raspian and chromium and make some configs for kiosk and now all is fine. The reason is - a new raspian use the modern x-server with all the possibilitys for unclutter cursor-hidding and so on.

iYaroslav, 2020-05-07:
In my case, I added a flag to start the X
```
startx -- -nocursor
```

closed 2020-08-24 by jareware