jareware/chilipie-kiosk #115: How to configure that the Monitor power off at 10 minutes and power on on touch?

blueicehaller, 2020-05-18:


codeclinic, 2020-06-17:
@blueicehaller Did you have to install xscreensaver? When I activate screen blanking via raspi-config I get the message "dpkg-query: no packages found matching xscreensaver".

blueicehaller, 2020-06-17:
Yes I confirm that behavior.

codeclinic, 2020-06-17:
So you did install xscreensaver?

blueicehaller, 2020-06-18:
No I didn't because it's not necessary in my case.
Today I tested on a fresh install and I just had to comment out two xset options and then sudo reboot.
You can test the behavior using 20 seconds:
`xset dpms 20 0 0`

Offtopic: Fresh install was necessary on a new SD Card because my previous no longer started with kernel panic.

bernardovalente, 2020-07-14:
@blueicehaller  I don't have `A6 Screen Blanking - Enable/Disable screen blanking` on advanced settings, any idea why? I just did a fresh install and for A6 is `A6 Pixel Doubling`.

blueicehaller, 2020-07-15:
Which model?
I am now trying to update my Raspberry Pi (I already upgraded raspi-config and currenty `sudo apt update && sudo apt upgrade` is running) to see if the option disappears.
At the moment the option is still available.

https://www.raspberrypi.org/documentation/configuration/raspi-config.md

bernardovalente, 2020-07-15:
3b+. It's working though, i just edited the .xsession file an installed xscreensaver. Thank you

codeclinic, 2020-07-16:
> 
> 
> 3b+. It's working though, i just edited the .xsession file an installed xscreensaver. Thank you

@bernardovalente  Please share your edited .xsession file. I also have the 3b+ and cannot get it working.

bernardovalente, 2020-07-29:
sorry for the delay:

`
xset +dpms
xset dpms 20 0 0`

blueicehaller, 2020-07-29:
Currently I don't have access to my Raspberries.
Can you please check if the Monitor power on if you call `~/display-on.sh`?
If yes maybe you can call it from the [gpio-watch](https://github.com/larsks/gpio-watch)?

bernardovalente, 2020-07-29:
yes i can call `~/display-on.sh` but i'm not really sure how to auto run the `gpio-watch -e rising 4` command on start uo or how to use a symlink, sorry i'm kind of new on raspberrys...

blueicehaller, 2020-08-24:
Did you meanwhile try to create an symlink `sudo ln -s /etc/gpio-scripts/4 ~/display-on.sh` ?

closed 2022-12-28 by blueicehaller